VoronDesign/Voron-Afterburner #6: Inconsistent extrusion / Z Banding

Roycinger, 2021-01-02:
Hi,
first and foremost I would like to bring this issue to attention where we can track it all together properly. Discord doesn't seem the right place for me as messages and focused discussions get overscrolled and interrupted with non-related topics very quickly.
Before I start I want to mention that I don't expect this issue to get solved by the Voron Team alone. Moreover, I want to evaluate this together with you so that the entire Voron community can benefit from it.

I started facing this extrusion issue when I completed my 100%-built-by-specc Switchwire. In the beginning I was thinking that the printer wasn't properly tuned yet. So I went through all the tuning steps (aligning belts, rails, tuning for filament, E-steps, run_current etc). The result didn't improve within about 90 or more test prints (where I tested every single extrusion related slicer and firmware setting).
Over the course of a few weeks, I partly reassembled the SW again, paying special attention for accuracy, squareness, etc, results didn't improve. At this point I was still under the impression that the issue was on my side as the issue still persists.
To rule out the AB, I converted my SW to bowden and the issue was gone! So I came to the conclusion that there might be a problem with the AB itself.
Because of that, I reprinted the Clockwork parts, changed the TL gears, changed all bearings, changed the motor, changed the entire hotend, precisely cut the PTFE tube. No change in results at all. 
For informational purposes, this is my config: https://pastebin.com/nyULd4ca
Fast forward:
Up to yesterday, I noticed that more and more people are facing the same or similar issues, even if they might be fine with the results.
I got sent a plethora of pictures which I was allowed to use in this Git issue. They all more or less show the same problem,
Please note that most pictures where shot with "bad" light, directly from the top to make the issue apparent.
These are kinda mixed from specc-built v2.4's and SW's.
[image]
[image]
[image]
Cubes from my SW:
[image]
[image]
Random v0 part printed with my SW:
[image]

For my part, I could rule out any looseness in the kinematics or topics related to filament (as I made tests with PETG, PLA and ABS from various brands). 
Those defects appear either random (not at fixed heights) or depending on print features within the model (plateaus, holes, etc).
 
For me it comes down to a few points:
- requirement of quality is not the same for every user (they might live with the defects)
- There might be issues with either the AB internat geometry or the ABS shrinkage bothering with the tolerances
- related to the point above: A genuine Bondtech BMG in a direct drive situation in another (non Voron) printer works perfectly without showing these defects
- Using a bowden extruder (M4 in this case) eliminated the issue

Yesterday, I had a brief conversation with Ray. I introduced the topic "Prusa Issue 602", which looks awefully similar to this. 
I would be happy if you could either assist with this issue or shed some light on it. 

Thanks in advance!

toskium, 2021-01-02:
Thank you for submitting this issue. I do experience the same results and I was not yet able to pin down the root cause of it.

toskium, 2021-01-02:
To add something to the issue. I was doing input shaper calibration and the outcome of the calibration print is imho interesting.

Since klipper calculates the extruder acceleration according to the xy acceleration the striping pattern changed accordingly.

See the attached picture.
[image]

It appears more or less intense starting from acceleration 1250 all the way up to 7500.

BladeScraper-Designs, 2021-01-02:
Same here as well.  tried new BMG parts, various flow settings, various accels and speeds from 30mm/s and 500acc to 120mm/s and 5k accel, etc.

[image]
[image]

Gizzzle, 2021-01-02:
Can confirm the issue.
[image]
[image]

mgrl, 2021-01-02:
Just built a v1.8 with bowden. I think I don't see the effect, which aligns to your proposal, right?

[image]
[image]
[image]

toskium, 2021-01-02:
> Just built a v1.8 with bowden. I think I don't see the effect, which aligns to your proposal, right?

Yes, correct. The issue seems to be exclusive to the clockwork extruder. Your input is much appreciated!

Serpreme, 2021-01-03:
Can we make a list of the items that are used and not used between the bowden and direct drive? The gears, motors, wiring, and then we can slowly eliminate those items. We can try to test for things like EMI for example

Gizzzle, 2021-01-03:
> Can we make a list of the items that are used and not used between the bowden and direct drive? The gears, motors, wiring, and then we can slowly eliminate those items. We can try to test for things like EMI for example

I can eliminate every part on the Afterburner except the hotend, but that's not the problem since the other guys are using a dragon while I'm using a v6. Also EMI is not the issue since I used another motor without cable management which lead to the cable not crossing another cable in it's way to the board.
Stepper driver was also swapped from a LV8729 to a TMC2209

BladeScraper-Designs, 2021-01-03:
Added details to my post.

jdlongenecker, 2021-01-03:
Hello everyone, I have a recommendation to make. 

In order make a proper comparison, I recommend that someone mount a BMG extruder directly to Afterburner in place of the Clockwork module. If the issue clears up, then you can root cause the issue to be something with the clockwork module itself. If the issue persists, then the source of the issue is something else.

I would not compare a bowden and direct drive mount of the same extruder--the physics of printing through a long tube with springy filament is different than when the extruder is right on top of the hotend.

Gizzzle, 2021-01-03:
> Hello everyone, I have a recommendation to make.
>
> In order make a proper comparison, I recommend that someone mount a BMG extruder directly to Afterburner in place of the Clockwork module. If the [...]

We had the idea too, but that's nearly impossible because of the small footprint of the afterburner.

JayAwesome87, 2021-01-03:
Same problem since day one. I'm using a Galileo Clockwork now and the problem still persists.

[image]

Roycinger, 2021-01-03:
> In order make a proper comparison, I recommend that someone mount a BMG extruder directly to Afterburner in place of the Clockwork module. If the issue clears up, then you can root cause the issue [...]

This is easier said than done because of a few points:
- The BMG is 2.6mm thicker than the Clockwork:
[image]
- The filament path is not the same as the Clockwork's:
[image]

All in all, almost every single part of the AB would have to be redrawn to accomodate a BMG

jdlongenecker, 2021-01-03:
> Same problem since day one. I'm using a Galileo Clockwork now and the problem still persists.
> 
> [image]

Is this the version with planetary gears? If so, this would imply that it's not the Clockwork module itself but something else :)

Gizzzle, 2021-01-03:
I'm using my Ender Switchwire Conversion

jdlongenecker, 2021-01-03:
> > In order make a proper comparison, I recommend that someone mount a BMG extruder directly to Afterburner in place of the Clockwork module. If the issue clears up, then you can root cause the issue to be something with the clockwork module itself. If the issue persists, then the source of the issue is something else.
> 
> This is easier said than done because of a few points:
> 
> * The BMG is 2.6mm thicker than the Clockwork:
> * The filament path is not the same as the Clockwork's:6bc29c552557.png)
> 
> All in all, almost every single part of the AB would have to be redrawn to accomodate a BMG

Let me see what I can do, I would like to support your testing! I do not have time right now for diagnosing the issue but I may be able to provide some parts you can print with to test the theory.

JayAwesome87, 2021-01-03:
> > Same problem since day one. I'm using a Galileo Clockwork now and the problem still persists.
> 
> Is this the version with planetary gears? If so, this would imply that it's not the Clockwork module itself but something else :) since there are no shared parts between the Galileo Clockwork (planetary) and the standard Clockwork BMG version.
> 
> Also, can we make sure to document which printer these examples are from? That would be helpful in the root cause!

Yep, the one with planetary gears. Printed on a Voron 2.4 250.

jdlongenecker, 2021-01-03:
[image]
[image]

I am thinking something simple like this--no part cooling, but it should still show improvement

Gizzzle, 2021-01-03:
I indeed worked on a solution like this. Need to wait until my V0 is ready again to print the parts...

[image]

jdlongenecker, 2021-01-03:
that looks good! I think it will work for testing 👍

herexx6349, 2021-01-03:
same issue hier with a brand new 2.4 with AB

[image]
[image]

herexx6349, 2021-01-03:
took some photos under a microscope, looks like a x/y wobble^^
[image]
[image]
[image]
[image]

Gizzzle, 2021-01-04:
Update: Adding the complete BMG didn't make a difference. So there must be a problem in something that the Switchwire and the V1/V2 have in common.

maximm07, 2021-01-04:
Just for my curiosity, my 2.4 voron is in final build stage so i will be able to test myself soon..is the issue appearing with all hot ends? dragon, V6 and mosquito? is it the same with any of the mods optimised for cooling (5015 variants)? current hotend cooling for dragon is not the best, for V6 it should be ok tough.

Roycinger, 2021-01-04:
> Just for my curiosity, my 2.4 voron is in final build stage so i will be able to test myself soon..is the issue appearing with all hot ends? dragon, V6 and mosquito? is it the same with any of the [...]

Afaik it happens across all hotends. I'm not too sure about the Mosquito as very few people are actually using it.

> Update: Adding the complete BMG didn't make a difference. So there must be a problem in something that the Switchwire and the V1/V2 have in common.

Please note that a BMG Clone was used by Gizzle for the test. 
In my initial post I was speaking of a genuine Bondtech BMG.

jtrmal, 2021-01-04:
@Roycinger how about trying "git bisect"? I'm in my post-construction afterglow and I don't have a good idea how my prints should/could look like, but bisecting might help you

Roycinger, 2021-01-04:
@jtrmal I dont see how a git bisect could help with this issue. Could you please elaborate?

jtrmal, 2021-01-04:
hm, I was thinking if you could pinpoint some working (i.e. producing nice prints) revision and using the current as "bad", you could (perhaps) manage to track down which commit has introduced this deterioration?

jtrmal, 2021-01-04:
but I just noticed there is an implied fact these are all new machines? In that case bisect wouldn't help, I think, yeah....

fkotey, 2021-01-04:
I have BLV MGN Cube and Ender 3 with the same issue as well. Currently, I am working on tuning my printers to print parts for a Voron Switchwire but this issue is preventing from doing that. The common thing between my BLV MGN Cube and the Ender 3 Linear Rail. I have been investigating this issue for months now and I believe it lies with the cheap linear rails (bought from Blurolls) I am using. 

I will confirm my assumption in a few weeks time as I have ordered genuine Hiwin linear rails to upgrade my printers. My Ender 3 with wheels never had this problem.

Roycinger, 2021-01-05:
As a general plea:
If you're experiencing a similar issue, please make sure it's really related to this issue.
Though it might look similar, the root cause can be vastly different.
The issue for me and @Gizzzle was solved by converting to bowden, so we can eliminate rails from as the root cause.
Rather, this strongly points to some aspect in the Afterburner

jdlongenecker, 2021-01-05:
Note there will always be some inconsistency like this--we can probably get a bit better to a certain extent but at harsh lighting angles there will always be some surface imperfection. Some reading that may be helpful is at this link (scroll down to the "inconsistent extrusion" section:

https://blog.prusaprinters.org/prusa-research-summer-update-2018_8535/

I still stand by my theory that bowden tend to not have this issue as much as direct drive (any direct drive) will. Any slight variation in the drive gears, filament thickness, etc--will directly translate into an almost 1:1 change in how much filament is pushed through the nozzle with a direct drive extruder. On a bowden extruder, all of those inconsistencies are dampened by the "spring" of filament through the bowden tube and the majority of that variation is smoothed out. Of course--bowden comes with other complications and downsides, but that's for another discussion 👍 

I think the part we are missing @Roycinger is a print on your Voron with a genuine BMG attached to the toolhead, similar to some of the issues above? If that resolves the issue we can point to Clockwork (or maybe the clone BMG gears), if it does not, then we know the issue is something else.

Roycinger, 2021-01-05:
@toskium thanks for uploading these detailed pics. It helps narrowing down the issue.
@jdlongenecker You are certainly right with the "filament as a spring" argument that smoothes out such imperfections.
Unfortunately, I don't have a genuine BMG attached to my printer. The examples with that genuine BMG I was talking about were from another (non Voron) printer that I already sold. I can cram out some example pics though sor sure.
For your edit: I have genuine Bondtech components on the way, along with SLS printed Clockwork parts. This way we can potentially eliminate 2 parts in the equation:
- genuine Bondtech gears vs. TL (or whatever brand) clone gears
- ABS / shrinking being the culprit.

I will keep you updated as soon as the parts get in.

raymondh2, 2021-01-05:
I run all genuine Bondtech parts in my Afterburner. While changing two variables (clockwork > BMG and TL guts to > Bondtech guts) might give insight I am highly doubtful in this. From what I have seen even the bowden M4 suffers from this too a degree.

ctilley83, 2021-01-08:
To me this looks like classic over extrusion. Probably by 3-5%.  Dial it back extrudermeister

hhammarstrand, 2021-01-08:
> To me this looks like classic over extrusion. Probably by 3-5%.  Dial it back extrudermeister

Mine is actually calibrated in superslicer. I can try lowering and posting the results later on, my v2 is currently undergoing maintainace

hhammarstrand, 2021-01-09:
[image]

Printed with my switchwire on 0.95 extrusion multiplier. My calibrated value is 1.03. 

Specs: 
Switchwire 
Bondtech gears 
Genuine v6, 0.4 nozzle 
LDO extruder motor

Roycinger, 2021-01-09:
Update from the initial post.
Re-aligned Z-Rails once again and re-did the belts.
[image]
[image]
[image]
[image]

Didn't upgrade to the genuine BMG gears yet as I'm still waiting for the SLS parts to be delivered

ctilley83, 2021-01-09:
> [image]
> 
> Printed with my switchwire on 0.95 extrusion multiplier. My calibrated value is 1.03.
> 
> Specs:
> Switchwire
> Bondtech gears
> Genuine v6, 0.4 nozzle
> LDO extruder motor

Looks a lot better.  Now just tune pressure advance and you should be good to go. I don’t trust calibrating using a 1 wall cube. Calipers aren’t accurate enough. What slicer are you using?

hhammarstrand, 2021-01-09:
> Looks a lot better. Now just tune pressure advance and you should be good to go. I don’t trust calibrating using a 1 wall cube. Calipers aren’t accurate enough. What slicer are you using?

Thanks for the input, but it's actually looking much worse overall. I'm using superslicer 2.2.53. Used superslicers built in calibration to do the extrusion multiplier tuning, not the "one wall cube".

9ary, 2021-01-11:
I haven't done enough testing on my SW to know if I'm affected yet, but I'd still like to pitch in because I find this issue interesting.
Just throwing some ideas at the wall here but some things I've noticed when building which might be worth investigating:
- The E motor shaft touches the filament above the hob gears, while it sounds unlikely there might be some sort of interaction there.
- The shaft the reduction gear and drive side hob gear are mounted to is supported axially by the motor faceplate and the wall in the extruder body. Ideally there should be a little bit of axial play or we could be facing unwanted drag. On the other hand, too much axial play there might cause the reduction gear to rub against the extruder body, and back and forth movement might actually explain why we're seeing inconsistency throughout the print. Really, the only parts of this assembly that should ever touch stationary bits are the outer bearing races. I also need to confirm whether one of the bearing can touch the hob gear or not.

The above points have caught my attention because the motor I was planning to use turned out to have an extra long shaft that's not compatible with the clockwork (it blocks the filament path), so I had to space it out. The reduction/drive side gear assembly has a lot of axial play because of that. I've noticed significant drag (more than expected) and possibly grinding while pulling filament through the extruder with the hotend removed. I've already got a new motor on the way and I'll report back as soon as it's installed.

My extruder internals and the motor in question are known good, I've used them to print all of my switchwire parts on my ender 3 and I haven't noticed anything like what you all are facing. They were part of TL's AL-BMG-AIR extruder kit.

rcfishhunt, 2021-01-12:
One issue that I did notice was the driven shaft slides side to side in the afterburner.   Mine has similar defects.   The gears are tl labs hotend pheatus dragon v1
[image]

hhammarstrand, 2021-01-12:
> One issue that I did notice was the driven shaft slides side to side in the afterburner. Mine has similar defects. The gears are tl labs hotend pheatus dragon v1

I noticed too. I used shims to remove the side to side movement and get the filament path spot on. Not sure if this has any real benefit, I guess the gears "auto align" when you put filament in.

9ary, 2021-01-12:
If you mean the idler, it's indeed supposed to self align when the filament is in. That's how it is in a stock BMG body too.

hhammarstrand, 2021-01-12:
> If you mean the idler, it's indeed supposed to self align when the filament is in. That's how it is in a stock BMG body too.

I was referring to the shaft assembly. The idler didn't have nearly as much play as the shaft assembly
[image]

9ary, 2021-01-12:
Alright, this might be worth looking into then, because it really needs to be constrained properly.

ctilley83, 2021-01-23:
Seems my afterburner is not affected.  You didn’t cut your ptfe tube too long did you? There was an issue on the prusa forums that if the ptfe tube gets compressed during assembly, it can lead to similar extrusion issues. The particular issue was with a prusa mini which has a similar geared setup to the afterburner. Seems that after properly trimming the ptfe tube, the issue disappeared.

ctilley83, 2021-01-23:
[image]

that’s the worst downlighting I could get.  Extrusion seems pretty spot on.  Only a few inconsistent spots around features

rcfishhunt, 2021-01-23:
I am not really sure.  I had to guess my length.   For the ptfe

JustinCaseOfWolf, 2021-01-23:
> I am not really sure. I had to guess my length. For the ptfe

i have cut the afterbuner top in half so i can measure exactly the ptfe length. not quite the fastest / cheapest way to do it but i know for sure it is the correct lenght. so in my case that does not apply

JustinCaseOfWolf, 2021-01-23:
@ctilley83 different colours shows different surface. while yours is less obvious than the black above i am sure with the right light your can be made more visible.

rcfishhunt, 2021-01-26:
I am making some progress with this.  And it might seem simple.   The build instructions say nothing about using the nylon washer on the thumb screw from the bmg parts.  I took mine off and replaced it with a thin flat washer to keep the spring from digging in to my printed parts and was able to reduce spring tension and it is making an amazing difference.   Still loosening to get the best results.   But her is with and without the nylon spacer
[image]

Gizzzle, 2021-01-27:
> I am making some progress with this. And it might seem simple. The build instructions say nothing about using the nylon washer on the thumb screw from the bmg parts. I took mine off and replaced it with a thin flat washer to keep the spring from digging in to my printed parts and was able to reduce spring tension and it is making an amazing difference. Still loosening to get the best results. But her is with and without the nylon spacer
> Uploading 20210126_174702.jpg…

Unfortunately I can't see your pictures, but I have used the washer from the beginning and my print quality was bad. So maybe it's only a coincidence in your case.

rcfishhunt, 2021-01-27:
Ok I will try to reload them

ehellman, 2021-02-04:
I have followed this discussion for a while and sadly haven't been able to chip in because I am still waiting for my POWGE package which contains all my pulleys etc.

Have printed the entire AB though, inspected, measured, tried V6 and Dragon variants. While I do have some ideas, first I would like a question answered, if anyone has time:

We see a bunch of calibration cubes here, great. But could someone take the time to create a vase mode print and post it here? Doesn't have to be a super advanced vase. I usually print a 100x30x120 (x, y, z) rectangle in vase mode to make some basic tests. Then I switch over to one of the widely available vase models, but the simplest ones are the ones that could show you the biggest effects. It should have some shapes, but not go overboard with it. Would be an interesting test at least.

mateuszbrzezinski, 2021-02-05:
I'm experiencing same issue, V2.4 with AB-BN-25 cooling
But how bad it looks it depends at what angle you look at and if the light is harsh or not.

[image]

mateuszbrzezinski, 2021-02-05:
What if that is some bug in Klipper? SKR? TMC2209?
Anyone using Duet to compare this?

9ary, 2021-02-05:
I've rebuilt my extruder with my new motor, and I'm not affected apparently. I'm using Klipper on a Duet 2 WiFi (TMC2660), nothing is tuned yet.

mateuszbrzezinski, 2021-02-06:
I was printing test cilinders, so there is no any sharp angles, so to exclude such variable. Then that is interesting, when nozzle crosses Y center axis of model, so before crossing this one of the AB drivers is rotating faster, then after crossing the other motor is rotating faster, then funny things happens, I marked point when that happens. So as you can see, on the left of that mark the issues are more visible, then suddenly after crossing the mark, on the right hand side, issues are much less visible, or almost not visible.
[image]
[image]

Maybe that is something with motor synchronisation?

mateuszbrzezinski, 2021-02-07:
Another experiment:
Printed 2 cubes like that:
[image]
You can see the difference.

Part printed at 90* angle front side:
[image]

Part printed at 45* angle:
[image]

But other interesting thing is part 90* but wall on Y axis are fine:
[image]

What the hell?

[17 more replies]

closed 2021-02-19 by raymondh2