VoronDesign/Voron-Afterburner #15: Afterburner heatset holes are 4.8mm instead of the 4.7mm used everywhere else.

ChaosBlades, 2021-09-27:
After building a v0 and working on a Switchwire mod I have noticed after printing the Afterburner parts a lot of the heatset holes are 4.8 mm instead of the standard 4.7 mm used everywhere else. This causes the heat sets to just fall into the hole and not actually melt any plastic. This seems to be a problem with all CAD for other printers that use the Afterburner as well. It is probably an issue with the STLs as well.

netweaver1970, 2021-09-27:
I did see in my MJF printed version of the afterburner AB-BN30 part cooling fan duct, that my heatsets were easily melting in, seemingly ok but had very little pull-out strenght. I blamed it on the MJF, as that was a fraction of percent bigger than an ABS part. But if it's indeed true about the slightly bigger holes, then I'm not surprised at all...

ChaosBlades, 2021-09-27:
Almost forgot. The ascent colored part cooling fan duct self tapping holes are also 0.1 mm larger than they are supposed to be as well. My screw just fell right in and didn't cut any thread. With that said I don't see why these are not heat sets as well. I modified my CAD of the Afterburner to put heatsets here.

0x3333, 2021-12-24:
This issue has been considered for the stealth burner release?

AaronHerrick, 2022-01-08:
> This issue has been considered for the stealth burner release?
Ihave checked SB, the only heat inset holes I have seen off spec in SB are on the probe bracket (4.8mm) and toolhead (4.6mm).  I think it is likely the ones on the toolhead were tightened by design.

qema36, 2022-02-08:
This seems to be an issue with the SB, I just printed a SB and the holes in the CW2 and SB are too big for the heatset inserts to grip into.  They don't quite fall into the hole but once pushed in they can turn in the plastic with a bit of force.   Same profile was used on a SW and I had no issues with the hole size.

closed 2022-04-21 by raymondh2