VoronDesign/Voron-Afterburner #16: Add hotend compatibility notes

andrhamm, 2021-11-11:
I'm new to the Voron community and was trying to determine which printhead parts to print for my Phaetus X Voron Dragon HF hotend. On Discord I was told to use the `TriangleLab_Dragon` models. I'd never heard of TriangleLab but this is apparently the same hotend made by the same manufacturer. A readme in the printheads directory might be a good place to put these kinds of notes where the models support multiple hotends or variants thereof.

LoganFraser, 2021-12-24:
The same factory makes the triangle lab and phaetus/BTT Dragon hotends. It's all the same except for the rectangle vs D shaped heat block

wile-e1, 2022-01-08:
This should be taken care of in the last pull-request that I submitted.

closed 2022-01-08 by wile-e1