VoronDesign/Voron-Afterburner #22: Holes in [a]_stealthburner_main_body_beta0.stl

elcrni, 2021-12-27:
Hotend cooling fan bay has 3 corners rounded and one left without fillet, this results in slicer having issues and the print has a hole in the wall. as seen on attached images
[image]
[image]
.

tnap1979, 2021-12-27:
[image]
Seems to be more of a cosmetic concern and less of a leak or performance issue.. but it's there none the less

FHeilmann, 2021-12-27:
Which slicer are you using? What's your "detect thin walls" setting?
Here's my slicing result with PrusaSlicer 2.4.0 and "Detect thin walls" enabled:

[image]

elcrni, 2021-12-27:
Also using Prusa 2.4, mz settings as attached
[image]

FHeilmann, 2021-12-27:
what are your extrusion widths?

elcrni, 2021-12-27:
All my extrusions are forced to 0.4mm

wile-e1, 2021-12-27:
This issue has been corrected with the update of [a]_stealthburner_main_body_beta1.stl

The missing fillet was added and material was also added to the front facets to help increase the wall thickness on this layer.

raymondh2, 2021-12-28:
Issue fixed in #21 Will leave open for three days before closing to confirm fix.

elcrni, 2021-12-28:
Is there a file [a]_stealthburner_main_body_beta1.stl available to download somewhere to test? many thanks

raymondh2, 2021-12-28:
https://github.com/VoronDesign/Voron-Afterburner/blob/sb-beta/STLs/Stealthburner/a_stealthburner_main_body_beta1.stl

closed 2021-12-28 by elcrni