VoronDesign/Voron-Afterburner #26: Phaetus BMS part cooling ducts are too high

DorjeDorf, 2021-12-27:
I just finished mounting a Phaetus BMS onto the SB. 
I noticed the ducts are high enough that they blow onto the heatblock - especially if a sock is present. This is very visible also during PID tuning. 

Photos attached. 
[image]
[image]

wile-e1, 2021-12-29:
Can you please measure from the top of the toolhead to the tip of the nozzle?

I check the CAD and everything matches what it should be.  I did, however, find an issue with the CAD provided by Phaetus that showed the radius of the heatsink being 3mm but in reality it is more like 1.5-2mm so I have a feeling that your hotend may not be seated fully.

plumpcheek, 2021-12-29:
Mine Phaetus BMS with beta0 toolhead and I measured from the tip of nozzle to the top toolhead (w/o ptfe collar) is 48.93 mm.
[image]
.

DorjeDorf, 2021-12-29:
In my case it's 51mm.

When you say hotend not being seated fully - do you mean the heatbreak in the heatsink or rather the whole thing inside of the printed part? If it's the latter - below is a picture to show the fitment. 

The one thing that has me curious is also visible on the photo from @plumpcheek - should a part of the heatsink be sticking out like that? My BMS is black and it's a touch more tricky to capture on a photo, but I see the same thing. 

[image]

plumpcheek, 2021-12-29:
Well, look like Phaetus strike again, my BMS is well within the back bracket, and you can count the number of heatsink fin, mine is 6 and ur are 7 :) This BMS purchased fom Trianglab as a drop in replacement for Ender 3 v2 hotend (that what Phaetus advertise), turn out it's not since its too short for origin E3v2 hotend fan duct, back then I remember I had to print a new duct for it, my wild guess is they extend the length to become a real drop in replacement hence yours is longer.
[image]

plumpcheek, 2021-12-30:
And may I ask what version of your back bracket, look like the corner radius is corrected with your hotend, if you used beta0 version, so there two types of BMS hotend: one is short and narrow corner and one is long and wider corner :)

DorjeDorf, 2021-12-30:
I used the Beta0 for CW2 one.

mathisschmieder, 2021-12-31:
I can confirm the 51mm length of a recently bought Dragonfly BMS. Well, more like 50.93mm

DorjeDorf, 2021-12-31:
Here are my measurements just to keep it in one place.
[image]
[image]

DorjeDorf, 2022-01-01:
I just printed out the Beta1 toohead. 

Things I noticed for the BMS. 

1. The name of the toolhead on the Front part - now reads BMO and not BMS. 
2. In my case, the change to the corners for mounting don't change anything, but are a nice touch :)
3. **No changes** in terms of the cooling ducts. Their bottom tip starts at the end of the heatblock when it's without a sock. When I put a sock on - it's clearly vislble that the air will blow on the heatblock and not on the nozzle. I did a water bowl test - there was center point for air converging beneath the nozzle. It got a bit better when I took the sock off, but still, nothing like what I can get with the Dragon SB toolhead.

AaronHerrick, 2022-01-05:
there are 2 different phaetus BMS toolheads. the older one, with 6 fins, is the one we did the mount for. the newer one, that they changed without telling anyone, has 7 fins and its slightly longer.  I'm hoping its about the same size as the Zodiac BMS, plus or minus .25mm that should be close enough

mathisschmieder, 2022-01-05:
According to these https://cdn-3d.niceshops.com/upload/file/Technical_data_sheet_for_BMS_Hotend[0].pdf specs, the Zodiac BMS is 45.36mm from top of heat sink to bottom of heater block. This is absolutely in line with my recently bought 7-finned Phaetus Dragonfly BMS and with what DorjeDorf posted.

mathisschmieder, 2022-01-06:
As far as I can tell, the beta2 parts fix this issue, at least with the 7 fin version. Top of toolhead to nozzle tip is bang on 61.5mm

raymondh2, 2022-01-07:
@DorjeDorf Does beta2 fix these issues for you?

closed 2022-01-11 by raymondh2

DorjeDorf, 2022-01-11:
Bah I missed the callout. I will print it tomorrow and let you know.

DorjeDorf, 2022-01-13:
Can confirm. Looks good. Haven't printed yet, but water bowl test went through great. Thank you.