VoronDesign/Voron-Afterburner #27: StealthBurner / ClockWork2 odds and ends...

Durahl, 2021-12-28:
So, I've now assembled my SB/CW2 Combo but I've yet to test it on my Voron 2.4 due to me still working on HartK1213 Toolhead PCB and an ERCF Toolhead Sensor compatibility - While the latter wasn't _that_ much of a problem the former still is - Kinda wish the Cable Door would offer like two more mm of free space inside:
[image]
I'll most likely have to modify the PCB by turning the PCF Board Connector into a Pigtail one 😑

**Aaanyway... A bunch of other things I've noticed along the ride that may also be of interest for the general public:**
1. **Manual Page 15** - The supposed M3x20mm screw should most likely be an M3x25mm too
( akin to the three other M3x25mm ones that hold the two CW2 halves together via Heat Inserts ).
2. **Part Interference** - The marked corners could use some trimming as they're interfering with the Idler BMG Teeth:
[image]
3. **Part Printability** - Continuing the split along the red marked line would significantly improve printability of the Cable Door as the small triangle ends up being a R.P.I.T.A. to print without it turning into a hot mess:
[image]

wile-e1, 2021-12-29:
> 1. **Manual Page 15** - The supposed M3x20mm screw should most likely be an M3x25mm too
>    ( akin to the three other M3x25mm ones that hold the two CW2 halves together via Heat Inserts ).

I put in a change request for the manual 2 days ago, so it should be updated soon.

> 2. **Part Interference** - The marked corners could use some trimming as they're interfering with the Idler BMG Teeth:

This has been addressed in my latest PR.
[https://github.com/VoronDesign/Voron-Afterburner/pull/38](url)

> 3. **Part Printability** - Continuing the split along the red marked line would significantly improve printability of the Cable Door as the small triangle ends up being a R.P.I.T.A. to print without it turning into a hot mess:

Unfortunately, this lower corner needs to be the way it is in order to make it easy to slot the hotend wires in.
If this portion is turning into a hot mess, Maybe try lowering your printing temp, increasing cooling, or increasing your threshold for layer speed reduction based on layer time.

raymondh2, 2021-12-30:
Will close once verified

closed 2022-01-21 by raymondh2