VoronDesign/Voron-Afterburner #28: More difficult to print than existing Voron parts

zapta, 2021-12-28:
Never had a problem printing Voron parts with my v2.4 but this one gave me unexplained artifacts. Lamination is good and the part is strong but there are very visible artifact lines that reflect features on the inside of the part. This is not a layer shift since that layer is OK on the rest of the perimeter of the part. 

I don't know if it's just me would would be useful to stay within the current requirements of Voron part printing.
[image]
[image]

evandepol, 2021-12-28:
@zapta i wonder if these are likely slicer-profile-induced artifacts. At first glance it seems you might have too high of a flow rate, since outside walls appear to be "bulging" differently when outside walls are thin (towards the bottom of the picture) versus thicker/more room for the filament to go.

zapta, 2021-12-28:
Erik, I am printing with Super Slicer, esteps calibrated and 0.95 flow. I
will try to recalibrate my printer and print again and will report here.

My point is that there is a potential that this model is more demanding
from existing Voron parts so it will be interesting to see if you will get
similar reports from others.

bartoszwas, 2021-12-28:
It's white it means that is layer separation due to shrinking.

Scroeder, 2021-12-28:
I have the same 2 lines. Interesting.

wile-e1, 2021-12-28:
While I admit that the SB main body is challenging to print, it **can** be printed without seeing these artifacts.  You're seeing these issues in these locations because of the change of cross-section there.

[image]

Do you have a chamber thermistor?   (if so, what temp is it reading?)
What material is this with?

Scroeder, 2021-12-28:
> While I admit that the SB main body is challenging to print, it **can** be printed without seeing these artifacts. You're seeing these issues in these locations because of the change of cross-section there. Also, there is a LOT baked into this package and you don't get the esthetics, LEDs, or cooling performance if the parts become extremely simplified. There may be some things that can be done to improve printability, but I won't be "dumbing" things down just so it's easier to print.
> 
> [image]
> 
> Do you have a chamber thermistor? (if so, what temp is it reading?) What material is this with?

I print with Esun Abs+ with around 40°C in chamber.   Imho print quality is perfect on other parts.

zapta, 2021-12-28:
The lines look white because of a lighting artifact, they are not white in
reality or when the light is from a different direction.

I was printing with Hatchbox black ABS, chamber temperature rises up to
45C. The print with the lines was done with no pre-heat soaking time, just
as I have done with all other Voron prints and never had a problem. I am
trying now with a 1h soaking time, to let things stabilize better, will
report the results here.

The part may be printable but if it raises the bar for Voron parts
prinability, that decision should be made consciously.  The problem is not
structural but aesthetic, at one of the most visible parts of the design.

zapta, 2021-12-29:
Here is the print with the 1h soaking. Overall its good but still a
deformation at the top of the part (as installed in the printer). The
profile should be made of two slopes but one if them is printed
concave (see annotations in pictures below).

[image: 2021-12-28_16-05-46.jpg]

[image: IMG-3212.jpg]

Scroeder, 2021-12-29:
I have found a picture on Facebook with the same strange lines. 
[image]

jarrettv, 2021-12-29:
My print had these lines too but I assumed it was more an adhesion issue. I think the surface area in this area is just smaller overall than AB/ABBN and makes it more likely for the part to pull away from the bed near the top. A small loss of adhesion in that area results in lines in the print. I'm using SuperSlicer too fwiw.

zapta, 2021-12-29:
After another examination, I believe that the problems in my print are due
to adhesion issues at the top end (as installed in a printer) of the part.

1. Examining the front face with a ruler shows that the top end was lifted.

2. If cut the model in the slicer and print only the top end, it comes
clean.

The structure of this part, long and minimal first layer area on one end
makes it more difficult to print than normal Voron parts. CW1 solved it by
splitting the face into two parts.

bartoszwas, 2021-12-29:
That's the problem. When adhesion is good there's risk it will crack like on zapta's picture (white lines). I printed it three times with ASA filament which I never had any adhesion issues (I've printed all Voron 2 parts that filament) and every time top part was lifted.

khay8485, 2021-12-29:
I had the same Issue with warping of the 5015-End of this part, which resultet in those thickened lines on the part. For me, changing filament did the trick. Changed from Fiberlogy ABS to eSun ABS+ to make it work. Surface was cleaned properly.

zapta, 2021-12-29:
This is a question to the designers of the new extruder.

Is it a goal for you that everybody that can print other Voron parts can
also print this one or is it OK to have additional
materials/printers/processes/skills requirements in order to print this
part successfully?

My preference is to prioritize printability and voron accessibility over
form.

SellA, 2021-12-30:
I think that these parts are more challenging than standard voron parts, but nothing compared to non-optimized parts. I would look from another perspective: if you are printing with a Voron, are you sure you are getting good enough prints from your printer? I've printed it in ABS and some tests in PLA and yes, i have some artifacts and small aesthetics issue, but i get those on all my prints and i'm still investigating it. Also my PIF parts that comes from 2 different machines (both from Stephan) got 2 different look: some are perfect and some shows layers bands.
So if this works as good as ABBN and got all that new features packed on a single toolhead, i think we must try to increase our printers quality and reliability. Maybe it's not easy but it will pay off on all other prints that we do.
Obviously if the part can be optimized i hope it will be done by developers.

zapta, 2021-12-30:
I reprinted the part with purple Elmer glue, bed adhesion adhesion was very good and the problem disappeared though the face is a little rough because of the glue, even after washing it out. This confirms that the cause is bed adhesion. 

Increasing the first layer surface will make this part more printable without scarifying functionality, e.g. by increasing the width from 3mm to 7mm in the picture below.  

[image]

elcrni, 2021-12-30:
my prints dont have any lines, as clean as it can be. turn on 'detect thin walls' as the lines are most likely due to lines overlapping.

hackerjack42, 2021-12-31:
I see the same line on my latest ASA print.  Textured plate....clean.....Prusament ASA....."beta 2" main body.  Beautiful print other than the line.  I checked the face for flatness with a straightedge and it did indeed lift at the top of the part cooling fan intake.  The contact patch is only about 3mm wide, which in my opinion is too small for an object that is intended to be printed in a filament that is known to shrink.  I'm sure that printing with a brim would resolve this, but it would be better to be addressed in the model itself.  Can that top contact patch be enlarged without compromising the design aesthetics?

to-the-nth, 2022-01-01:
I agree these parts are harder print.  I've printed 3 sets of Voron 2.4 parts in eSun and Paramount ABS with a 51-55C chamber (Elmer's Xtreme glue stick slurry on FR4 110C bed -way better than Purple). I got occasional lifting on the first set of parts, so now I just print with a brim and have had -0- failed prints.  Brims are a pain to clean up, but post results are excellent in ABS.  No lifting issues printing StealthBurner with a brim.  But a Stealth test print without a brim failed spectacularly. I am not a fan of eSun ABS+.  Colors are off.  ABS+ heat distortion temp is only 73C compared to 78C for ABS, and it does not clean up well if you have to use a brim or otherwise trim.  Prusament ASA looks interesting but limited color selection.  Amazon has Polymaker ASA might be worth looking at. We need to error on the side of being able to do consistent prints for the masses.  Scary when PIF parts have lifting.

I had bridging issues on the left duct in the pic below.  It eventually did complete the bridge, but it left a deposit of spaghetti underneath while trying.  I ran some test prints, but with the 55C chamber, there's not enough cooling and stuff just sags, so on the print below I resorted to using supports in Simplify3D.    Tried painting support in SuperSlicer and disabled no support under bridge, but couldn't get it to work. Wonder if it is getting confused because the base surface is contoured and not flat.

[image]

Durahl, 2022-01-02:
I may have missed someone mentioning it but [those lines](https://user-images.githubusercontent.com/2184769/147517747-0ea26853-f266-4405-8410-63f3154b649c.png) are caused by the **Top Layers** that are now also being generated at those Z-Layer Heights causing both the Z-Height generation to slow down but also contracting thus resulting in the newly generated layers above those areas to appear where they're supposed to while the previous layer underneath them have noticeably shrunken down.

I started my Voron journey on a Prusa i3 MK3S using **Generic ABS** ( couldn't get the much revered ABS-Plus at the time ) but have since changed over to **Dutch Filament ASA-X** that has solved a lot of my woes I had with the ABS by simply printing balls to the walls fast **_and_** at full fan speed ( not possible on the ABS I used ) which I assume cools the already laid down filament fast enough before it can add up for enough force to curl up on me.

[In this video](https://youtu.be/HPOSy8ITktM) you can see the finished Toolhead in motion under changing light conditions and as a still under harsh light conditions where you can _ever so slightly_ see the same two lines if you know where to look for:
[image]

You _could_ try printing those areas without a top layers taking up time to mitigate the issue? 🤔

garyt72, 2022-01-04:
I had a similar issue printing the SB on my first attempt.  I "renewed" by PEI sheet by going over it lightly with ScotchBrite, washed thoroughly with with soap and water, then dried and cleaned again with IPA before letting the sheet sit on the heated bed for ~30 minutes to dry completely.  I haven't had an issue with any the 3 SBs (and other items) I've printed since then.

AaronHerrick, 2022-01-08:
Using a new build plate solved my issues as well.

closed 2022-01-21 by raymondh2