VoronDesign/Voron-Afterburner #29: Chain anchor mounting tolerance/suggested improvement

evandepol, 2021-12-28:
When mounting "chain anchor_3hole-beta0.stl" to "motor_plate-beta0.stl", I put the heatset insert in a tad too deep (perhaps 0.5mm or so). This was enough for the m3x20 BHCS screw to not engage with the thread. So it seems the designed bolt length is a tad too short in some cases.

When i used an m3x25 instead, I blew out the back of the part while tightening.

Suggestion to make this more robust/less critical:
- update instructions to use m3x25 instead of m3x20.
- modify the "chain anchor_3hole-beta0.stl" to include an m3 relief opening so that the longer bolt can go in without bottoming out inside the part.

[image]

evandepol, 2021-12-28:
Per discussion in discord, this has been resolved by user wile.e by changing the bolt relief on the chain mount.

raymondh2, 2021-12-28:
Leaving open for three days before closing

closed 2021-12-31 by raymondh2