VoronDesign/Voron-Afterburner #30: [SB-beta] guidler tight fitment / suggestion to allow for more tolerance

evandepol, 2021-12-28:
It seems the idler gear assembly is really tight when sandwiched between the motor plate and other half of the CW2. 

The cw2 bolts are pretty loose right now (finger tight) to avoid binding, and it still seems there is not a lot of play. I sanded down the idler assembly to give it more play and it's better now, but it seems pretty tight in there. Can the two plastic standoffs holes that meet the motor plate be taller (in 0.4 -0.6mm range perhaps?), or can the holes in the motor plate be made slightly less deep, or can the idler assembly be trimmed in order to give the assembly just a little bit more play? I think the assembly would be more forgiving it had a little bit more tolerance and had easier swing.

Perhaps also of note is that i print on textured plate, which may affect thickness tolerances as well (only on the idler assembly though, since the insides of the CW2 sandwich would not be affected by that). The same printer and plate was used to print Voron 0.1 parts, and that went together easily, so i doubt my 2.4 setup is that far off.

I used motor_plate-beta0.stl, main_body-beta0.stl, [a]_guidler_a-beta0.stl, and [a]_guidler_b-beta0.stl

SellA, 2022-01-01:
I also have some struggles with the tolerances on the guidlers. It seems that they are not all the same measure and too tight also for a press-fit. I attach some sectional view of the beta 2 STL. I think that these tolerances should be made parametric to test it out and try to find the perfect fit.
[image]
[image]
[image]

wile-e1, 2022-01-08:
The guidler has an adequate amount of clearance built into it in order to function as intended without binding. The guidler needs to have support on both the front and rear faces in order be fully constrained and prevent twisting from the eccentric loading on the idler.  Having a large amount of clearance would make assembly much easier, but could also come at a detriment to the performance of the extruder.

If the guidler is this tight for you, I suspect you may be over-extruding slightly.  The tolerances on CW2 were tightened up quite a lot in order to improve performance.  Also, you may want to see if you have any blobs or overlap bits that may be causing a tight fit-up.

closed 2022-01-23 by wile-e1