VoronDesign/Voron-Afterburner #31: [SB-beta] Phaetus BMS fit issue with toolhead rear

plumpcheek, 2021-12-28:
File print: stealthburner_toolhead_(phaetus_bms)-rear-cw1-beta0 (and front), the upper corner radius seem wider than the hot-end corner radius lead to the upper end of hot-end not sit flush and the bolt misalignment with hole below (a little, I can force it in). See picture below:
[image]

wile-e1, 2021-12-29:
The CAD provided by Phaetus does not match actual parts.

[image]

The radius of the model shows 3mm, but measurements from actual heatsink are around 2mm.  The toolhead will be adjusted to suit.

plumpcheek, 2021-12-29:
Thanks, closing this and wait for new toolhead.

closed 2021-12-29 by plumpcheek

wile-e1, 2021-12-29:
Should be posted tomorrow.