VoronDesign/Voron-Afterburner #33: Phaetus BMO part cooling ducts are too high

Krieselwind, 2021-12-28:
distance between nozzle and highest point of fanduct is about 3mm.
[image]
[image]

raymondh2, 2021-12-28:
Duplicate of #26

Krieselwind, 2021-12-28:
40.4mm

[image]

44.9mm
[image]

11.6mm
[image]

raymondh2, 2021-12-28:
Can you please check and make sure your screws aren't bottoming out.

AaronHerrick, 2021-12-28:
I can confirm that the BMO I used to test with is the same overall length.  I would guess that either the screws are bottoming out for you, or something else is preventing the hotend from seating fully.

Krieselwind, 2021-12-29:
Screws are tight. My second BMO was still in use and have the same size.

mathisschmieder, 2021-12-29:
I can confirm both the measurements and the issue. While the 8mm M2.5 screws supplied with the hotend in fact do just bottom out on the heat sink, it is seated as far up as the printed part allows. Using 6mm M2.5 screws - while definitely preventing bottoming out -  leaves dangerously low thread engagement. 

Top of plastic part to the tip of the nozzle measures 61.6mm:
[image]

wile-e1, 2021-12-29:
The nominal design height for the top of the toolhead to the base of the nozzle is 61.5mm for all hotends in Stealthburner, so you're pretty damn close at only being 0.1mm longer.

The airflow is aimed downward at the nozzle, not flat across the nozzle, so I think this may be more of a perception that the nozzle is too low.  My BMO looks exactly the same as yours and it's working ok, but I will admit that when the silicone sock goes on the heatblock, it looks like the airflow to the nozzle may be getting blocked slightly.

I think this warrants a little more investigation.  For now, I have revised the toolhead to reduce the possibility of the supplied fasteners being too long.

AaronHerrick, 2021-12-31:
the problem is me.  I've been saying the stickout should be 1.7mm, I was wrong. it should be what it is:  ~2.4mm

raymondh2, 2022-01-07:
@Krieselwind Can you please try the latest parts and let us know if that fixes the issue for you?

Krieselwind, 2022-01-08:
I see no difference?!

[image]

AaronHerrick, 2022-01-08:
Please measure the distance as shown here: https://github.com/VoronDesign/Voron-Afterburner/issues/33#issuecomment-1002460699

Krieselwind, 2022-01-08:
61.7mm

AaronHerrick, 2022-01-08:
61.7 is fine.  The spec distance is 61.5mm.  different printers and plastics and measurements can account for that 0.2mm difference.

This mount is correct, the picture looks correct, and it should work correctly.  I made a mistake telling you that the stickout measurement should be 1.7mm.  the correct measurement for that should be around 2.4mm.  please refer to https://github.com/VoronDesign/Voron-Afterburner/issues/33#issuecomment-1003238959

It is better to measure from top of toolhead to nozzle (61.5mm)

I think we can close this issue now.

Krieselwind, 2022-01-08:
Ok, fine ☺️ thank you

closed 2022-01-11 by raymondh2