VoronDesign/Voron-Afterburner #34: San Ace B52 does not fit

Scroeder, 2021-12-28:
I have a problem mounting the B52 fan. Had to dremel out the screw hole, but it seems that 2mm in diameter are missing. 

[
[image]
[image]
](url)

wile-e1, 2021-12-29:
I have added a little clearance to the fan cutout for SB Main Body based on BadNoob's feedback on his San Ace fan  I do not have a B52 to test fitment though.  If you could print Beta2 and report back, that would be fantastic.  Thanks!

Scroeder, 2021-12-29:
Sure, where do I find the beta2 ?

wile-e1, 2021-12-29:
My PR isn't merged yet, but you can access it from here:

[https://github.com/VoronDesign/Voron-Afterburner/pull/38](url)

Scroeder, 2021-12-29:
Unfortunately it does not fit but definitely better. I removed a little material from the marked area. Now it fits so nice and flush. 
[image]

[image]

[image]

raymondh2, 2022-01-08:
We have decided not to support this fan due to the B52 being a 52mm fan not fitting into the space constraints in the design. While it can be forced into place, we neither recommend nor condone this method of fitment.

closed 2022-01-08 by raymondh2