VoronDesign/Voron-Afterburner #35: Unsupported brackets - Cura exclusive slicing tolerance

kamermans, 2021-12-28:
When printing the `main_body_beta1` in Cura with `Slicing Tolerance: Exclusive`, the supports for these overhangs (the green parts) are not printed at all, so the blue sections are unsupported spaghetti:

[image]

Here's the toolpath view in Cura:

`Slicing Tolerance: Exclusive`:
[image]

`Slicing Tolerance: Middle`:
[image]

This is the first issue I've encountered with `Slicing Tolerance: Exclusive` - I've printed an entire Voron 0.1 and a few Voron extruders with this enabled.  If `Middle` should be used, we might want to mention that in the docs somewhere.  I can put in a PR if that's the route you want to go (assuming the docs are on GitHub).

Settings:
Cura 4.12.0
Layer height: 0.20mm
Filter out tiny gaps: enabled (disabling didn't help)
Fill gaps between walls: everywhere

kamermans, 2021-12-28:
Here's the finished print.  There are 4 spots where this occurred (blue) and one that printed fine (yellow). 

[image]

wile-e1, 2021-12-29:
I can understand how this is an issue with the settings you have shown.  Unfortunately with how thing the bridges are intended to be, you've found the scenario where they do not work.  In general, I think most people wouldn't be touching settings like that unless they had an understanding of the consequences.

The only way to really get around this is making the trim-away bridges thicker which could create other issues for most people who use "middle".  I can look into the issue a little.

I'd be interested to understand why you're using a different slicing tolerance.  Could you elaborate as to why?

wile-e1, 2021-12-29:
I also wanted to bring this to your attention that the slicing tolerance is listed in the "Experimental" features section of Cura..

[image]

I don't think there's much more here that I can do since this seems to be more of an issue with an experimental setting on Cura rather than a design concern.

kamermans, 2021-12-29:
Ah, good point, somehow it escaped me that this was experimental.  I was using it to print Voron parts on my Ender 5 which was not well-tuned and was overextruding on corners, making fitment of functional parts needlessly difficult.  It is a nice compromise in most cases since it doesn't lead to underextrusion that I might otherwise get if I reduced the flow.  Clearly this is a problem of tuning (acceleration, jerk, etc), and the fact that it's also experimental makes me totally content with closing this issue.  If you do need to tweak the bridges in the future, note that the top bridge printed perfectly since it ends up with two layers instead of one.

closed 2021-12-29 by kamermans

kamermans, 2022-01-07:
Just wanted to note that `slicing_tolerance = exclusive` is set as default in the official Voron 2 Cura slicer profile:
https://github.com/VoronDesign/Voron-2/tree/Voron2.4/slicer_profiles/cura/Voron_Cura