VoronDesign/Voron-Afterburner #37: SB clearence with Rapido strain relief

DaLeberkasPepi, 2021-12-29:
My Rapido Hotend came with an included strain relief. Which i know the early review units didnt have. But thats also was causing problems for me. With the included strain relief i cant get the Rapido to fit into the cable channels.
Printed Parts:
[Stealthburner_toolhead_rapido_rear_cw2_beta0, Stealthburner_toolhead_rapido_front_beta0]

[image]
[image]

My suggested solution would be to make a cutout at the side of the cable channel (like shown in the picture below)
[image]
[image]
[image]

raymondh2, 2021-12-29:
This is already being worked on

raymondh2, 2022-01-07:
@DaLeberkasPepi do the latest files fix the issue fo you?

DaLeberkasPepi, 2022-01-07:
I'm in the process of reprinting that piece. I'll post pictures of it in place but the new version looks good.

DaLeberkasPepi, 2022-01-08:
With the new version the rapido fits no matter what orientation it gets mounted in. 

At the beginning i didnt know that i could rotate the heatsink when i removed the 3 screws (I did think it would have been a too tight press fit or something) so i could have made it work by rotating it maybe... sry. 

But honestly i dont want anyone encourage to remove the 3 skrews to rotate the heater core only to tighten them wrong and bending the stainless steel heatbreak....

TLDR: Now it should work no matter how oriented your heater block gets mounted on rapido. Thanks wil.e and the Voron team for the great work!

[image]
[image]
[image]
[image]

closed 2022-01-08 by DaLeberkasPepi