VoronDesign/Voron-Afterburner #42: IGUS chain mount does not line up with cable hole

RealDeuce, 2021-12-30:
With the IGUS chain mount, the strain relief tab extends past the edge of the cover.  As a result, the zip-ties on the strain relief tabe get pushed back toward the chain, and the wires all need to make a sharp 90° turn to get under the cover.

[image]
[image]

RealDeuce, 2021-12-30:
Using just one screw to move it back by the hole spacing helps somewhat (the cover is a lot easier to close), but it's not quite enough to allow the string relief to work properly.

raymondh2, 2022-01-07:
Can you please post some pictures that are a bit clearer about what's going on.

raymondh2, 2022-01-07:
I now understand the issue. Are there any reasons other than the wires making a sharp 90 degree turn? @RealDeuce

RealDeuce, 2022-01-07:
> Can you please post some pictures that are a bit clearer about what's going on.

If I had an extra chain end, I could mock it up, but as it is, I won't be able to take further pictures until the next time I take the CW2 apart... but I'll try describing the issue better.

With the IGUS chain ends, there's a strain relief tab that extends from the end of the chain end.  The purpose of this is to provide a place where you can use zip-ties or similar to bundle the wires and ensure they don't slide in and out of the chain when the chain flexes.

If you have a zip-tie at the end of the strain relief tab around the wires, that zip tie holds the wires in a position that is past the opening in the back of the cable cover, which means a wire exiting the end of the chain and properly attached to the tab needs to go backward toward the chain before it can enter the opening for the wires under the cable cover.

If the wires are properly attached to the tab, closing the cover will force the wires to make this bend and hopefully slide the zip-ties back along the tab.  With two spec zip-ties, this forces all the wires to make a tight 90° bend to enter the opening.  Three zip ties forces an even tighter bend.  If the mount holes on the arm are moved back so the tab ends approximately even with the edge of the stepper, the wires can do a nicely radiused bend into the opening under the cover and still be held properly by the tab.

On my setup, I initially put the end hole of the chain end into the second hole on the arm, moving it back about 7mm, which was a large improvement, but still not quite enough.  I ended up modifying the STL to move them back a few mm more, which provided for a nice bend in the wires.

raymondh2, 2022-01-07:
Closing as the design team does not see this as an issue.

closed 2022-01-07 by raymondh2

RealDeuce, 2022-01-07:
> I now understand the issue. Are there any reasons other than the wires making a sharp 90 degree turn?

It's actually even more than 90° if the wires are properly held at the end of the tab, and even when using two zip-ties at the base of the tab, the bend is being made by closing the cover, which puts strain on the cover and hinge screw (and wires of course).

Since the wires are likely hidden from view while closing the cover, it may not be clear that the wires are the cause of the extra resistance and they may end up bent tighter than their then minimum bending radius, making a wire break more likely to be caused by opening and closing the cover.