VoronDesign/Voron-Afterburner #45: LED carrier clear plastic unnecessary

richardjm, 2021-12-31:
I'm using natural ABS as my diffuser and led carrier but in an attempt to let more light through I drilled a hole in the carrier. This does allow a little more light through and I think negates the need for the carrier to be made from clear plastic.

Just a thought.

[c]_stealthburner_LED_carrier-beta1

[image]

timmit99, 2022-01-03:
What natural ABS are you using? Im using some Stratasys Natural ABS and my LEDs need to be set to 0.6 max to get a good glow. Any BRIGHTER than that and it starts to become hard to see the logo.

wile-e1, 2022-01-03:
The part is designed to be printed with clear/translucent material. If users print the parts as intended, the current design will function as intended.  This seems better-suited as a UserMods once CAD has been released.

If you're printing a diffuser out of clear/translucent material, printing the carrier in the same material at the same time makes sense.  The thickness of the portion you drilled out is only 0.8mm thick, so I don't know if it's going to make any appreciable difference by removing it.

closed 2022-01-03 by wile-e1