VoronDesign/Voron-Afterburner #46: StealthBurner x_frame_V2TR_MGN12 - LH & RH

MartynJ970, 2021-12-31:
Hi, 
Can the top bolting location for the two bodys be moved up 0.8mm, Reduce the Depth of the Counter Bore,Hex.
Also the 3.3mm that the CW2 be all the way through.

This will then allow the Modified CW for the ERCF to fit on the assembly?
I have donthe LH side however the RH stl file has gaps and cannot be repaired in SW,IN,F360

images.githubusercontent.com/74456725/147823126-f5d5afca-0f2c-4fc4-a786-19673db896ca.JPG)
0

Durahl, 2022-01-03:
You can make the changes in F360 by simply modelling the changes in a separate mesh and then import the original mesh together with the separate mesh into a Slicer and either ( to remove material ) have the separate mesh act as a modifier with 0 Perimeters, Top, Bottom, and 0% Infill or ( to add material ) leave it as normal mesh.

It is how I added support for my **Mosquito Magnum**, a cavity for the **ERCF Tool Head Sensor** and mounting hardware for the **HartK1213 Toolhead PCB** to the SB & CW2:
[image]
Still had to modify the PCB by turning the PCF and HEF Board Connectors into Pigtailed ones - Also disassembling the CW2 is again a R.P.I.T.A. as the Toolhead Sensor is blocking one of the screws that would normally be accessible to dismount the CW2 as a whole from the X-Carriage.

Will most likely mod the crap out of it once we have access to the CAD Solids 😏

raymondh2, 2022-01-06:
Closing as this seems to be for a mod and not the core files

closed 2022-01-06 by raymondh2