VoronDesign/Voron-Afterburner #49: Blower Intake Not Centered

to-the-nth, 2022-01-01:
Latest Beta2 STL. Front view shows blower STL intake not centered to fan or body of STL.  Using a Delta BFB0524HH from Filastuder which appears to be authentic.  There are a lot of counterfeits out there.  Fan is not getting loaded quite as much as with the factory housing: 124mA @24V with original cover vs 114mA mounted in STL so the fan could be loaded more... 

Back view shows fan well centered left to right. Hard to measure. Fan may need to be moved down relative to the top or intake hole moved upwards depending on design intent.

I would expect the intake hole size to be slightly greater than the inside impeller blade diameter?  I measure a 32.25 mm dia intake hole on the blower cover that was was removed while the STL is ~30mm.  There also is a ~3mm gap between the impeller blades and the intake hole while distance to the original fan cover is ~2mm.  The larger gap which could reduce efficiency? What is the thinking for how large the gap should be?

Looking forward to the release of the step files to allow local quick iterations.  Thanks for all the work!

[image]
[image]

raymondh2, 2022-01-01:
We are already aware and looking into it.

wile-e1, 2022-01-03:
The industry 'standards' for 5015s are about as loose as it comes with respect to placement of the impeller and dimensions of housings/mounting holes.

I was asked to offset the cutout for the fan inlet slightly, but this may not have been a good idea.  We're gathering data to try and figure out how best to address the issue.

to-the-nth, 2022-01-03:
I hear you.  I've modeled a blower for a DIY printer. The problem becomes way more complicated trying to create a single universal design optimized for all fans.  Two suggestions:

-Pick a few fans to support and create separate  STL's if necessary.  We do it for hot ends.  Hope the vendors don't change their tooling. Might make life easier in the long run.

-Design the CAD so it is modifiable.  I edited the AB-BN version 30 to adjust the spacing between the fan and the impeller.  Unwinding all the fillets and chamfers in the right order to get to a base editable core that I could start modifying was daunting.  In the end of course there were some simple fixes, but it took a lot of energy to get there.  Shout if there is anything I can help with.

evandepol, 2022-01-03:
Suggestion: while I like the idea of the integrated clamp, but given that there are already holding screws as part of the design in the present, it might be more pragmatic to have a number of separately-printed “blower fan shims” as individual and customizable parts.

These shims might be a suitable semi circular or crescent shape, that replaces the current clamp, be affixed with the screws, and could accommodate and be customized to the various 5015 fans. 

It would essentially separate this part of the geometry from the stealthburner main body  geometry. Much easier for people to print and experiment with shims, than trying to provide and maintain multiple iterations of the SB main body.

AaronHerrick, 2022-01-08:
Please check with the Beta 3 stealthburner carrier. while we cannot completely eliminate offset we can reduce it.  I also have found that using the FHCS in the back, ou can slightly adjust most fans, but this adjustment is limited.

closed 2022-01-21 by raymondh2

Eclypse3D, 2022-10-13:
I had the same problem with a Sunon Motor is off to the right of the housing by about 2.5mm but my Delta is spot on dead center.  I do not think it will affect much if at all however, but in my all white clear stealthburner its super obvious how off it is, as opposed to my copper one but there is not getting around it, if you try clearance anything in the cover you will reduce airflow or seal.

Headrush69, 2023-01-22:
> I had the same problem with a Sunon Motor is off to the right of the housing by about 2.5mm but my Delta is spot on dead center. I do not think it will affect much if at all however, but in my all [...]

I had the same issue with a Sunon MF50152VX-1L02C-A99. So much for spending extra on a "better" fan. Swapped in a cheap Amazon GDSTime fan and much better alignment but still slightly off. Checked all dimension and printed Stealthburner is perfect. 

With the Sunon fan, fitting is very tight. (Sanded down completely) 
Would this alignment affect functionality or just be a cosmetic problem?

mWachu, 2023-03-13:
Same issue with Sunon fan and latest stealthburner release. This really sucks... 
[image]

VoronManiac, 2023-03-13:
Part of the problem is there is no standard for blower fan dimensions.  Each manufacturer has their own footprint.  I think the Voron team has done a great job trying to find a common denominator. The design was been updated since my original post.  I am printing up the very latest STL's and will report back if we not in pretty good alignment.

Also stay away from under powered fans like the Sunon and stay with BB fans.  I highly recommend getting an authentic Delta BFB0524HH which will set you back under $20. Delta is rated at 160mA or 3.84W (easily measured)  where the Sunon is about 1/2 the power at 1.94W.  There are MANY fake versions of the Delta fan out there both from China and US sellers.  Confirm with the seller will stand behind the markings on the label.  I've had good luck with GDSTime GDB5015 which comes in at 100mA @24V for other applications and are available for $8 on eBay, but again, wouldn't recommended a lower powered GDSTime fan for this application.  The Delta is VERY loud running full tilt.  You can dial back the power in your slicer settings, but it is good to know you have the airflow for things like bridging.

mWachu, 2023-03-13:
> 

Would you mind to show how this Delta fits? Looking for good 5015, currently got a 12v sunon but it has a huge offset from the center. Too much for my aesthetics :D

VoronManiac, 2023-03-14:
FYI I'm at 5000ft.  Just measured current to be 130mA @24V on BFB0524HH across four fans I got the fans from Filastruder.  The fan is pretty well centered side to side, but it does ride a touch high.

[image]

[image]
[image]

raymondh2, 2023-03-17:
This won't be fixed. Stealthburner was designed around the delta fan we recommend. It's not going to be perfectly centered on other fans

VoronManiac, 2023-03-17:
Form the photos I posed, I think it is looking good with the Delta fan.  Nothing to fix. The delta seems to be one of the more powerful fans out there for the money.  Great design choice.  Delta's are available from reputable 3d printer suppliers at least in the US and were also available from electronic supply houses like Digikey and Mouser at one point.  Just stay away from the cheap ebay counterfeits. Measure the current to know that you have a genuine Delta.