VoronDesign/Voron-Afterburner #50: Probe mount screws extend into back of Rapido

xLAwildx, 2022-01-01:
The screwheads on the PINDA probe mount extend into the back of the Rapido tool head mount back.

xLAwildx, 2022-01-01:
[image]

wile-e1, 2022-01-03:
You're using a non-standard probe on the carriage.  

If you can provide some info on the mount you're using I can look into accommodating things, but I can't make any promises.

xLAwildx, 2022-01-03:
It's the part known as pinion.stl in the switchwire release. It perfectly
fits the PINDA2 probe, so i thought it was officially supported.

wile-e1, 2022-01-03:
> 
 You're absolutely right, this is supported.  I'm not familiar with Switchwire at all, so I didn't realize that's where this was from which is why I revised my initial comment.  Sorry for the confusion.

wile-e1, 2022-01-03:
[Pinion-Revised v1.zip](https://github.com/VoronDesign/Voron-Afterburner/files/7800769/Pinion-Revised.v1.zip)
Can you please try and print this part and let me know if it works for you.

Also, I would highly recommend BHCS for these locations.  They take up much less height.

raymondh2, 2022-01-07:
@xLAwildx Do the files Wile.e provided fix the issue for you?

xLAwildx, 2022-01-07:
Printing the probe mount just now, but the stl looks good.

xLAwildx, 2022-01-08:
[image]
Screws on the probe mount are now verified to sit below the front edge of the x carriage. Problem solved.