VoronDesign/Voron-Afterburner #51: Probe intersects with switchwire x carriage

xLAwildx, 2022-01-01:
PINDA style probe intersects with switchwire x carriage. There is a contour on the linear rail mount, but the opposite contour is missing from the x carraige.
[image]
[image]

stvptrsn, 2022-01-05:
Hi xLAwildx, the pinda probe has a really short, like 2.2mm, sense distance.  At the point the probe is interfering with the x carriage, like in your pictures, it is too far off the bed to trigger.  I just confirmed this on my Switchwire with the pinda probe installed
[image]
.

nife87, 2022-01-08:
I agree that its sense distance is short (I have a genuine SuperPINDA), but having just converted from AB v2.4r1 to SB beta 3 I disagree that the clearance is large enough. My probe is sitting 1.4mm above the nozzle, and there was not enough space, so I had to grind the corners a bit. See #57.

xLAwildx, 2022-01-08:
[image]
You can see here where I manually carved out part of each side of the x carriage so that I could move the probe high enough.
[image]
You can see in the second pic that this way I can move the probe mount up to level with the bottom of the carriage, which should be the max height.

wile-e1, 2022-01-21:
This issue should now be resolved with PRhttps://github.com/VoronDesign/Voron-Afterburner/pull/67

closed 2022-01-21 by wile-e1