VoronDesign/Voron-Afterburner #52: Bottom heat sink fin on V6 is not properly cooled.

Dfdye, 2022-01-05:
The bottom heat sink fin on the E3D V6 heat sink is not enclosed within the area cooled by the tool-head fan.  The V6 heat sink measures 42.7 mm from end to end, and 39mm from the upper inside edge of the mounting groove to the bottom of the lowest cooling fin.  The SB mount for the V6 measures approximately 36 mm from the upper edge of the mounting groove to the bottom ridge of the cooling duct that encloses the heat sink (about 3mm short of what is needed, if my measurements are correct).  This leaves the lowest fin of the heat sink un-cooled, and in my case seems to have caused heat creep issues for me while printing PETG,  I don't have definitive evidence for this, but anecdotally the PETG will not experience heat creep when fitted with the ABBN.  As a point of comparison, the ABBN V6 mount appears to fully enclose the V6 heat sink within the area cooled by the tool head fan.

AaronHerrick, 2022-01-06:
this is certainly possible, what heat break are you using?  if it is one with a teflon liner, have you recently changed it the teflon liners oxidize over time, even at low temperatures.

Dfdye, 2022-01-07:
I tested with both a Ti heat break and a bi-metal Ti/Cu heat break.  Both were "all metal" and did not have tubing in the heating zone.  It may well be that I have something else going on (which is why I added the caveat in the initial post!), but so far it seems to be that the cooling from the SB is not working exactly right for my rig.

raymondh2, 2022-01-07:
We are looking into this

Dfdye, 2022-01-07:
Great!  Thanks for all of the work that you have already put into this.  You and the rest of the crew working on this are fantastic!

wile-e1, 2022-01-08:
A revised toolhead has been created and will come out with the next round of updates. Expect to see in soon.

Dfdye, 2022-01-08:
You guys are unbelievable!  Thanks so much for looking into it.

govido, 2022-01-15:
[image]
[image]

I can confirm that it is fixed with beta2 :)

closed 2022-01-15 by raymondh2