VoronDesign/Voron-Afterburner #56: Issuing an Upgrade Warning

Durahl, 2022-01-08:
I think the Dev Team ( or the person curating the Manuals ) may want to add an Upgrade Warning/Notice that the new SB/CW2 combo will both change and _reduce_ the build volume in most if not all axes.

The MGN12 Carriage is ~4mm wider than the AB/CW one ( increasing the chance of a collision when attempting to reach `X0` on a 350 sized setup ), the Nozzle is being pushed like 3mm towards the front of the build volume ( necessitating the move of both the Z-Probe _and_ Build Plate ) and the increased height of the entire SB/CW2 assembly _reduces_ the Z-Build Volume height by 15-20mm.

AaronHerrick, 2022-01-08:
For X axis, are you using the same xy joints from the dual MGN-9 setup? the MGN12 requires new XY joints.  I don't know if that was explained in the manual.  Once you are using the correct xy joints you should have no issues in X or Y

For Y axis: The nozzle is in the exact same position as Afterburner.  the clearances in Y are in the exact same position.

For Z axis: here is a comparison of Afterburner to StealthBurner, this is approx 8mm.
[image]

wile-e1, 2022-01-08:
> I think the Dev Team ( or the person curating the Manuals ) may want to add an Upgrade Warning/Notice that the new SB/CW2 combo will both change and _reduce_ the build volume in most if not all [...]

Thanks for bringing this up.  The MGN12 carriage is 3.2mm wider than the standard MGN9 carriage, but the XY mounts were adjusted to accommodate this.  The cable chain contacting the top of the XY joint cap would and start to bind before the X carriage contacts the XY joint anyway.

As for the difference in Y that you mention, I'm not sure where you're getting this data.  The current CAD vs. the SB shows me that they are the same.  From the center of the X-rail, 91.7mm to the front face of the toolhead.

Current AB on MGN9 x2
[image]

Stealthburner on MGN12
[image]

As for the Z-height issue you mentioned, CW2 extends 2.7mm higher than the heights point on the existing Afterburner/CW1 design (which is the guidler latch).  However, the first point of contact (and the limiting part) is actually the Y-chain anchor that would contact the rear Z Idlers or the underside of the frame extrusion first.  So the different in height of the toolhead is inconsequential.

closed 2022-01-08 by wile-e1

Durahl, 2022-01-12:
Sorry the late reply...

> As for the Z-height issue you mentioned, CW2 extends 2.7mm higher than the heights point on the existing Afterburner/CW1 design (which is the guidler latch). However, the first point of contact (and [...]

Yeaaa... Pretty sure the Bowden Tube, you're probably forgetting about, would like to disagree with _that_ assessment:
[image]
Any higher than `Z280` and I'm expecting an `M117` with a restraining order from its lawyer to pop up in my Console.
[image]
The difference may just be like 10mm but... Ya kno'?

So yea... 330mm
[image]
...have long since left the chat ( even with the AB/CW my previous `printer.cfg` had to have Z already capped to 300mm ) 🤔