VoronDesign/Voron-Afterburner #59: Two sharp corners broke layers while fast printing

adama119, 2022-01-09:
[image]

Since CW2 is beta, I've just try to use  Stealthburner on my My Voron2.4  with Clockwork 1 and Phaetus  dragon hotted.
But the Led corner touch layer while printing some times.   Sometimes layers are broken by the corner.
Can you fix this?

stealthburner_main_body_beta2.stl
stealthburner_toolhead_(dragon)-front-beta0.stl
stealthburner_toolhead_(dragon)-rear-cw1-beta0.stl

raymondh2, 2022-01-09:
The way you make it sound seems like your prints are warping into the ducts? Can you please elaborate on this issue.

AaronHerrick, 2022-01-09:
At all times, that corner is 2.3 mm higher than the nozzle unless there is something broken on your printer.

stantang986, 2022-01-10:
It happened to me as well when using Voron-Dragon HF Hotend only. 
This hotend seems a lit bit shorter than normal one.

adama119, 2022-01-10:
> It happened to me as well when using Voron-Dragon HF Hotend only.
> This hotend seems a lit bit shorter than normal one.

Exactly the same,  My hotend is  Voron-Dragon HF Hotend

stantang986, 2022-01-10:
If you don't want to change stl file, 
you can try to screw the toolhead front part first and then cover the main body part(the out red one), it may save you just a little bit distance. 
Or, don't mind looking thing, jst cut it...

wile-e1, 2022-01-11:
I took a closer inspection to this issue and it does have some merit.  I have been using the Dragon mount and definitely noticed my toolhead is close to the build plate, but it still shouldn't be wiping your prints out unless they're warping or peeling up from the print surface.

With that said, the cutout that I was provided looks like it places the nozzle about 1.4mm higher than the target height, which absolutely does eat into the clearance. The Dragon mount has been adjust and will be updated on the repo soon. Please keep an eye out for Beta1 version of the toolhead files.

Thank you for bringing the issue to our attention.

vjeko2404, 2022-01-13:
I can confirm that with the dragon hotend and SB, the nozzle tip was only 0.7mm higher from the SB casing. My solution was to print 1mm distance washer for inbetween HE mount and the body.

closed 2022-01-15 by wile-e1