VoronDesign/Voron-Afterburner #71: [SW-Beta1] Dragon ducts are too high after recent model update

DorjeDorf, 2022-01-23:
Hi,

I recently printed out the newest Dragon mounts (beta2 front and beta1 cw2).

Previously I had the beta0 for both.
I noticed that now, the ducts are very high - to the point, where the air is blowing on the sock and significantly impacts the PID profile - to the point of triggering heater fault when the fan kicks in. This was not the case with beta0. Now, they are as much of a problem as we've had the BMS mount before in ticket #26 

When looking at the underneath of the mount - the bottom plate of the Dragon is sticking out of the mount - which was not the case with the previous version. 

To be sure - are the socks not causing an issue here? If I take them out, just visually I can tell that air will not really impact the heatblock. But with the sock on - it seems as if some of it would be blocked the sock. I don't have a way of running a flow simulation, so I may be talking out of my ass here, but... I'm worried this is still an issue. 

I'm attaching pictures below of before and after. 

New ducts: 
[image]
[image]

Old ducts:
[image]
[image]

DorjeDorf, 2022-01-23:
This is the TL version - bought a looong time ago ;) 
Top of heatsink to nozzle: 46.7mm
Top of heatsink to heatblock - 41.35mm
Top of heatsink to heatblock with sock - 43mm

wile-e1, 2022-01-23:
I'm honestly not sure what is going on here.  The mount is designed to what all of the current specs are showing for dimensions (45.5mm).  The question is, why is your dragon longer than what the specs sheets say (and what other users have confirmed with their hotends)??  Your Dragon is longer than it should be. (by about 1.2mm)

Specs from Phaetus show the Dragon is supposed to be 45.5mm from top of heatsink to nozzle.  The Triangle Lab info i found also shows that.

The toolhead is design for 45.5mm.

[image]

[image]

[image]

Is it possible that you're using a non-standard V6 nozzle?

DorjeDorf, 2022-01-23:
The nozzle is from Trianglelab - a T V6. It looks different to a standard E V6, but lengthwise is the same (both are 12.5mm). Just to be sure - I exchanged it to a regular V6 type - problem remained, since the problem is visible also without the nozzle. The distance from the top of the heatbreak to the bottom of the heatblock is different than what you show. 

The Dragon is very old - one of the first ones after they got released (it still has the V1 heatbreak). Maybe (like with the BMS) - something changed over time. 

That said... Looking at that bottom image (with a blue dragon) - it seems there's more going wrong with my mount. The bottom of the dragon on your picture sits pretty much flush with the bottom of the mount - notice in my pictures - mine sticks out quite a bit. 

Just to be sure, I'll reprint the mounts and check. 

Can you just confirm one thing for me, please? 
Does the sock not impact air flow? Looking at the bottom image - it seems as if there is no sock present there - not sure what was used for sims - and it's bugging me a lot ;)

zachdthompson, 2022-01-26:
Dragon v2 here, I just wanted to chime in and say I’m not having this issue. I built mine yesterday and it looks fine.
 
[image]

Im fairly certain this may be due to you having a v1 dragon. I tested both my standard flow and high flow dragons and they both fit.

DorjeDorf, 2022-01-26:
Actually, looking at your print @zachdthompson - I see the bottom of your Phaetus Dragon (judging by the heat block) protrudes underneath the bracket just as it does for me. This does not match what @wile-e1 posted in CAD. 

I should finally be able to print my set today. Will update then.

alexeypetrenko, 2022-01-26:
Hi, I'd like to get some clarity regarding dragon mount as well.

Should the bottom of a heatsink be flush with the round cutout on the toolhead front and rear parts?

In my case it sticks out around two mm.
[image]
[image]
[image]
[image]
[image]
These are `beta1` parts. I'm printing front beta2 right now to test it. But it is mostly irrelevant, as this distance beta1 and beta2 parts is the same (24.5) mm

[image]

Also, it looks like the CAD cutout from [this post](https://github.com/VoronDesign/Voron-Afterburner/issues/71#issuecomment-1019569990) does not match STL parts and dragon heatsink dimensions.

Stl leaves 24.5 mm space for 26 mm heatsink. So, it should stick out 1.5 mm. Thickness of a heatsink is only 2mm in the thinnest part. So, it should stick out at least 3/4 of it's thickness. But on a cutout is sticks out just slightly.

Sorry if i'm being picky. Just trying to understand how it's supposed to be.

DorjeDorf, 2022-01-27:
I made another mount print - no changes for me, but I see others are also seeing the heatsink stick out from underneath the mount effect.

Caipitrooper, 2022-01-30:
Hi,
after trying the new Dragon beta2 mount, I found the same issue on mine.
So I imported the Dragon Front beta0 and beta2 into Fusion and compared them.
The measurements show, that the Hotend mount on the **toolhead went down 1.43mm** (beta0=14.57mm, beta2=16mm) while the overall dimensions of the tollhead stayed the same. The lower surrounding structure even went further up (beta0=40.57mm, beta2=40.50mm). This moves the Dragon 1.43mm down blocking more of the vent outlets and leaving more of Dragon's cooling plates outside the cooling window (1.5mm less of the cooling body).

Here are the images I took in Fusion360,
the Dragon Phaetus SB toolhead beta0:
[image]

and the Dragon Phaetus SB toolshead beta2:
[image]

AaronHerrick, 2022-02-03:
I think this may be from using longer nozzles.  Those of you with the issue can you post pics of your hotends with measurements with and without nozzle?

DorjeDorf, 2022-02-03:
To be clear - in my case, the nozzle is exactly the same length as expected regular V6. This data I have already provided above. There is a chance that I just have a different dragon from everyone - it would be helpful if someone with a V1 chipped in, since I think most people have a V2 (and that one has differences in heatbreak length and the radiator itself too - in V1 - the heatbreak sits flush against the heatbreak, while V2 has a small part pushed in, so the heat break goes a bit into the top plate). 

That said, there seems to be a second potential issue where the radiator sticks out in the bottom - which does not match the CAD image.

AaronHerrick, 2022-02-03:
I have a trianglelab Dragon HF purchased in March 2020...this should be a v1.
It matches wile's and trianglelab's dimensions. 26mm heatsink, 45.5mm top of heatsink to nozzle

AaronHerrick, 2022-02-03:
I'm trying to reconcile all the data here, so one thing I did was try to pull some dimensions from your photo. 

that looks OK to me in terms of overall length, did you make a measurement along the blue line here?
[image]

DorjeDorf, 2022-02-03:
Ah ok. If your heatbreak is like mine and it sits flush against the radiator (no hole in for the heatbreak to go into) - then yes, you have a V1. The main difference is in the thickness of the heatbreak - it goes along what Slice did with their heatbreaks, where v2 (as can be seen in the posting from alexeypetrenko) received an almost full length copper wrapping, whereas in v1 - it's only the radiator fins at the bottom). 

The distance in my case for the blue line is actually 62.7.

Is your heatsink also sticking out from the bottom?

AaronHerrick, 2022-02-03:
its an HF:
[image]

Bastlwastl84, 2022-02-08:
I had tested the Beta and having the same Problem. Also that the standard Flow Dragon will getting to hot and i get heatcreaps. The nozzle is nearly 2mm deaper (the Vinda Sensor touches the Bed) than the basic Afterburner and the ABNB.

hmouridsen, 2022-02-12:
I have the Phaetus Dragon HF Voron edition, and I don't think I have the same issue, as shown here - it doesn't look that bad?

AaronHerrick, 2022-02-12:
> o that the standard Flow Dragon will getting to hot and i get heatcreaps. The nozzle is nearly 2mm deeper

can you please provide the measurements of your dragon? Measure: 
top of heatsink to tip of nozzle
top of heatsink to bottom of heat block
top of heatsink to bottom of heatsink

and finally, measure top of toolhead (part the dragon is mounted in) to nozzle

much appreciated.

What version of the beta toolhead are you using?

AaronHerrick, 2022-02-12:
> I had tested the Beta and having the same Problem. Also that the standard Flow Dragon will getting to hot and i get heatcreaps. The nozzle is nearly 2mm deeper (the Vinda Sensor touches the Bed if changing back to ABNB and AB) than the basic Afterburner and the ABNB.
can you please provide the measurements of your dragon? Measure:
top of heatsink to tip of nozzle
top of heatsink to bottom of heat block
top of heatsink to bottom of heatsink

and finally, measure top of toolhead (part the dragon is mounted in) to nozzle

much appreciated.

What version of the beta toolhead are you using?

cpfarhood, 2022-02-12:
> I have the Phaetus Dragon HF Voron edition, and I don't think I have the same issue, as shown here - it doesn't look that bad?

Currently printing a piece with a 1.25mm additional clearance, I’ll post some pictures when it’s done.

cpfarhood, 2022-02-12:
> I'm honestly not sure what is going on here. The mount is designed to what all of the current specs are showing for dimensions (45.5mm). The question is, why is your dragon longer than what the [...]

My dragon is in fact 45.5 top of heatsink to bottom of nozzle. I measured both the stock dragon 0.4mm nozzle and a 0.6mm bondtech cht, they’re within .1mm.

K4r98, 2022-02-16:
Having the same issue here with my Phaetus Dragon HF. Air is blowing against the socket. If I run my blower at 100%. Temperature drops a lot and print fails. Can only run at 35/40% Fanspeed. I think the height is fine. It's more the angle of airflow but can't fix it without CAD files :D

AaronHerrick, 2022-02-16:
the airflow angle is the same for all hotends.  can you 

> Having the same issue here with my Phaetus Dragon HF. Air is blowing against the socket. If I run my blower at 100%. Temperature drops a lot and print fails. Can only run at 35/40% Fanspeed. I think the height is fine. It's more the angle of airflow but can't fix it without CAD files :D

The airflow angle is the same for all hotends. can you provide a picture and measurements as described above?

cpfarhood, 2022-02-16:
Could this be related to the holes that feed from the part fan duct to the hotend fan duct?  I'll need to test with a disassembled piece but it seems like the real problem is cooling on the ceramic insulator not the heater block if all these dimensions are correct.

AaronHerrick, 2022-02-17:
What are your PID onstants, and what wattage is your heater?  do you run that with or without a sock?

> Having the same issue here with my Phaetus Dragon HF. Air is blowing against the socket. If I run my blower at 100%. Temperature drops a lot and print fails. Can only run at 35/40% Fanspeed. I think the height is fine. It's more the angle of airflow but can't fix it without CAD files :D

AaronHerrick, 2022-02-17:
> Right, my thought was that could turn in to suction with the part fan running, providing additional cooling to the hotend. It's easy enough to test but really grasping at straws.

that can't really happen. not the way the duct is shaped. 

The answer here is most likely more mundane.  When a problem occurs like this its common to look at the duct and think the answer is its blowing on the block.  After all, that's what is new to you, that would explain the data, so it seems reasonable.

But the real answer must explain ALL the data, including all of the other dragon tests.  there are hundreds of people using SB with Dragon (or its predecessor ABBN) with no issues.  If your dragon is the right height, and the duct works well for lots of others, there is another reason.  here are two I have seen:

1)PID_Kd term is too high/PID tuned with fan off.
Klipper's PID tune is very simple and can lead to bad answers.  if the Kd term is high (above 120 or so) that can lead to poor control

2)No sock/ heater underpowered
The heater might be up to the increased airflow 

3)too large a smooth_time is used for the thermistor.  I think Klippers default is 1-2 seconds, you really only need .25 -.5 s of smoothing.

DorjeDorf, 2022-02-17:
It's difficult to have a constructive conversation here, where most of our questions are ignored. 

Considering I started this thread, I'll say this.

1. My current status is I have bought a new dragon - V2. Now the lengths are correct. My guess is the problem with my V1 was the length of the heatbreak, but I won't be buying another one just to confirm this. 
2. The heatsink sticks out from underneath the mount - multiple people have asked if this should be like that - I can't see an answer to these questions here, despite those being asked loads of times. 
3. Now regarding the questions asked in the previous message, in my case - I am on a Duet and not on Klipper. My pid was done with the fan both on and off - duet calibrates both and has two curves for PID. I am using a sock - and actually I asked loads of times trying to understand where the issue is - if the sock should be on or not - no answer to that. Heater in my case was 50W before, now it's 60W - no change in behavior, but even a 40W should be more than enough. There is no smooth_time configurable on duet as it is in klipper (or I don't know of it). 
4. None of these issues were present on AB-BN. It just worked for all of my hotends and worked really well. 
5. Honestly, at this point - after having my questions ignored multiple times and seeing the thread was marked as invalid from the getgo - I won't be spending time here anymore. I appreciate the work that was put into the SB, but I don't appreciate the way our questions are being ignored. I mean how difficult is it to say - yes, it's weird the heatsink sticks out from the mount - we'll look into it; or no - this is the way it is going to work, thanks for noticing. Anything...

AaronHerrick, 2022-02-17:
DorjeDorf,
I'm sorry you feel that way.  For me, I am much more used to working on the discord server.
If you like I'd be happy to talk with you there either in a public discussion , ticket discussion (which only you and team members can see) or DM.  my user name is "Badnoob" and I designed the ducts and thermal performance of both SB and ABBN.

I apologize if if it seems I've been ignoring your questions. The bottom of the Dragon "heatsink" is really just a frame.  95% of the cooling is done by the small copper  finned section.  Its hard to say whether that frame should "stick out" or not, because that is a subjective term, and doesn't really impact the performance.  Your nozzle should be 61.5 mm below the toolhead top.  If that measurement is correct, and you are using the correct and latest sb beta files, you should be getting good performance.

Yes you should be using a Sock.

Without knowing the actual PID terms you have or having more data (such as cooldown curves with and without fan) I can't be much more help to you.

K4r98, 2022-02-20:
PID Tuning done with Fan on.
Socket on.
I bought a 50w Heater and it's able to hold the temperature :)

Airflow could be more downforced instead of blowing against the Socket but in my Case I am able to print with 100% Fan  :)

raymondh2, 2022-04-30:
Closing from inactivity

closed 2022-04-30 by raymondh2