VoronDesign/Voron-Afterburner #78: No room for the collet on a Pheatus BMS

WanderW, 2022-01-31:
Hi, I'm trying to build a SB with a Pheatus BMS (6 fin version) I previously used this hotend in an ABBN-30 and it worked great there. In the SB it doesn't fit, because there is no room for the collet that holds the PTFE tube. 
Because there is no room for the collar, the screwholes don't line up with the heatset inserts.

Stealthburner:
[image]

For reference the same mount from ABBN-30:
[image]

DorjeDorf, 2022-01-31:
Remove the collet - that's what I did. It won't be necessary considering the pressure coming from both halves of the mount that hold the tube in place.

timmit99, 2022-01-31:
Keeping the collet installed, while keeping the hotend stock, can make removal of the PTFE impossible without a full disassembly. Also, you would want to fit a collet clip in there as well\ to keep the tube from moving up and down during retract moves.

Removing the collet entirely and just having a spare one incase it gets damaged and you want to reinstall later is probably the easiest solution unless you see a NEED to have the PTFE captured inside the toolhead.

raymondh2, 2022-02-02:
Closing as we will not be making space for the collet.

closed 2022-02-02 by raymondh2

raymondh2, 2022-02-02:
After discussion, removing the collet, while potentially damaging to the collet, allows for much easier replacement of the PTFE and servicing of the hotend. In this case we decided against adding the cutout therefore requiring the collet be removed.