VoronDesign/Voron-Afterburner #84: Incorrect screws in manual for front plate

jdknoll, 2022-02-08:
The last page of the manual specifies two m3 x 25 screws for the front plate on clockwork 2, but it should be m3 x 20mm or else the screws stick through pretty far.

wile-e1, 2022-02-08:
> The last page of the manual specifies two m3 x 25 screws for the front plate on clockwork 2, but it should be m3 x 20mm or else the screws stick through pretty far.

The page where the bolts are showed also has a big note next to it explaining this.

[image]

timmit99, 2022-02-08:
@jdknoll  the screws are designed to be a bit long to allow for them to be loosened for toolhead removal without coming out of the heat-sets entirely.
=)

jdknoll, 2022-02-09:
Thank you for the response! With the highlighted note, my assumption was that you would loosen them slightly as in just enough to loosen them, but not remove them - not that you would leave them in the insert as you swap out, so when I saw the screw go past to where the guidler is, I assumed that something was wrong in my assembly. That said, thank you again for the clarification.

closed 2022-02-22 by raymondh2