VoronDesign/Voron-Afterburner #90: Rapido mount is oriented the wrong way

riley-williams, 2022-02-16:
The current rapido mounting orientation blocks the outlet on one side of the back piece:

[image]

It looks like the opposite rotation was intended as that would hold it up against the front and out of the way of the back outlet.

[image]

DaLeberkasPepi, 2022-02-17:
If you carefully remove the 3 stabilizing screws, you can rotate the heat block. But be careful when tightening the 3 screws again to go carefully and bolt them evenly, so you don't bend the heatbreak.

riley-williams, 2022-02-17:
Is this something every rapido owner will be expected to do? If all rapidos come in 1 or 2 orientations, it seems like it would be much easier to just provide the parts that support the factory orientations.

riley-williams, 2022-02-17:
After playing with it for a little bit, I've changed my mind. Seems like a reasonably easy thing to ask everyone to do. I'll track down a set of official instructions from phaetus on safe reassembly and create a PR for a README that might help with future confusion for people that get their rapido in the 2 unsupported factory orientations

wile-e1, 2022-02-17:
Is your Rapido a Phaetus or from some other vendor?

riley-williams, 2022-02-17:
Mine is a TL Rapido

nathanielhudson, 2022-02-18:
If you want another datapoint, I also have a TL Rapido and had the same factory orientation as riley. Disassembling and reassembling to correct this wasn’t too bad though.

wile-e1, 2022-02-25:
Since we've added the note for now, thanks to @riley-williams , I'm going to consider this issue closed for the time being.

closed 2022-02-25 by wile-e1