VoronDesign/Voron-Afterburner #91: Dragon hotend is loose even when screws are fully tightened

mathiasm74, 2022-02-17:
The four M2.5 screws that came with my HF Dragon have 12 mm thread. Fully screwed into the hotend, 7 mm thread is still visible, but the holes in the Afterburner are only about 5.5 mm deep, meaning the hotend can move up and down almost 2 mm even when the screws are fully tightened.

raymondh2, 2022-02-17:
Try using some M2.5x8

dmpmassive, 2022-02-17:
The wholesale Dragon's that come in kits like the Formbot seem to ship with M2.5x12mm and the Retail Box has M2.5x8mm.

mathiasm74, 2022-02-18:
Yeah, found my regular flow Dragon I bought earlier, and it came with 8 mm screws. Have modified the STL in Tinkercad now, but a lot of people must have the same problem.

geterdonek, 2022-02-18:
This is noted in the sourcing guide.

hmouridsen, 2022-02-18:
My Phaetus Dragon came with shorter screws meant to be used with V0 - they fit absolutely perfect

wile-e1, 2022-02-19:
While it may be noted in the V0 sourcing guide, it could certainly be more visible elsewhere.  I just don't know what the looks like yet.  But using 8mm long bolts is the solution to this "issue".  We cannot control what hardware ships with vendors kits.  All phaetus hotends that I've received generally come with a few different lengths of bolts. 

We will figure out the best place to document what's needed for the builders.

closed 2022-02-19 by wile-e1