VoronDesign/Voron-Afterburner #99: main_body-beta7 possible weak point

wattsie, 2022-02-22:
Hi All,
  Just started printing and playing with Clockwork2.

  I mock fitted the leavers to the "main_body-beta7", and while it might have been my heavy hand, the top guide section snapped while testing the leaver action. (currently printing another while trying to clean up my print quality)

  Note: I didn't have the motor plate attached, so understand it wasn't fully supported on both sides.

  However, wondering, as there is some additional space between this guide and the leaver, would it be worth adding some more support to the top of this guide to add strength?

[image]

Possible additional support location?
[image]

  Many thanks,

Cheers.

wile-e1, 2022-02-22:
This issue has already been sorted out prior to this post.  The parts are being test-printed but have not yet been posted.  Please stay tuned for the next update.

wattsie, 2022-02-22:
Thanks @wile-e1 ill wait to test next version.

closed 2022-02-22 by wattsie

wile-e1, 2022-02-22:
> Thanks @wile-e1 ill wait to test next version.

Posted in SB Feedback channel.

This space didn't open up until the last revision of the latch and I wanted to test that first before changing anything else.

[image]