VoronDesign/Voron-Afterburner #102: Missing perimeters in stealthburner_toolhead_(v6_r6)-rear-cw1/2-beta3

Promix953, 2022-03-04:
[image]
[image]
[image]

wile-e1, 2022-03-05:
This issue has been corrected in PR #103 

Thanks for the bug report!

Promix953, 2022-03-05:
> This issue has been corrected in PR #103
> 
> Thanks for the bug report!

[image]

The left hole still has problem

closed 2022-04-03 by Promix953