VoronDesign/Voron-Afterburner #110: stealthburner main body causes crashing in simplify 3d

Balinerd, 2022-04-08:
when trying to slice the stealthburner main body in simplify 3d at 0.2 mm layer height with 4 walls is causes a crash.

Steven23456, 2022-04-08:
I've sliced the beta 6 in SuperSlicer 2.3 (a fork of Simplify 3D) & Cura 4.13 at .2mm with no issue.

dmpmassive, 2022-04-08:
SuperSlicer is a fork of PrusaSlicer.  Simplify 3D is an older closed source slicer.

Balinerd, 2022-04-08:
> > when trying to slice the stealthburner main body in simplify 3d at 0.2 mm layer height with 4 walls is causes a crash.
> 
> can you link the file "stealth burner main body" ... I can't find for try with my S3D version.

Under stls

https://github.com/VoronDesign/Voron-Afterburner/releases

CrazyCreator13, 2022-04-08:
[image]
[image]

No problem

raymondh2, 2022-04-09:
> > > when trying to slice the stealthburner main body in simplify 3d at 0.2 mm layer height with 4 walls is causes a crash.
> > 
> > 
> > can you link the file "stealth burner main body" ... I can't find for try with my S3D version.
> 
> Under stls
> 
> https://github.com/VoronDesign/Voron-Afterburner/releases

Those are not the latest STLs. Please read the notes on that page

CrazyCreator13, 2022-04-09:
It's not about whether it's up-to-date. it's about the fact that he couldn't slice them with s3d

raymondh2, 2022-04-09:
That does matter. As the old slicing the old STLs doesn't tell us if the current ones have issues.

CrazyCreator13, 2022-04-09:
But I don't find the newest version ... 
or rather: I just wanted to show the questioner that it is not because of the stl. not more

raymondh2, 2022-04-09:
I have talked with some Simplify3D users on the Voron team and none of them have ran into issues. I suggest you contact Simplify3D support for assistance. https://www.simplify3d.com/contact-us/

closed 2022-04-09 by raymondh2