VoronDesign/Voron-Afterburner #119: Non-connectness of inner/outer walls leads to more seams

grmocg, 2022-04-25:
With the current StealthAB model (currently Beta7), the inner and outer walls of the SB are likely to be connected via infill, and this leads to seam lines on the outer walls of the SB. More icky than needed.

If we added a small void between the inner and outer walls of the SB, we'd be likely to trick slicers into connecting the inner/outer walls by running a wall along the sides of the void, which would lead to "moar prettier prints".

Wall == Perimeter.

grmocg, 2022-04-25:
Am referring to the "Main Body" piece, which is the one that arguably has the largest cosmetic impact.

raymondh2, 2022-04-25:
How many walls are you using? You shouldn't be seeing the infill connections.

grmocg, 2022-04-25:
I've done 4 prints at 3 wall.

I'm not seeing the infill connections, I'm seeing the seam starts for the layer(s), and observing that if the perimeters were connected from the outer-to-the-inner, there'd be half as many seam starts.

grmocg, 2022-04-25:
[image]

Looking at seam lines like the one highlighted in this screenshot

raymondh2, 2022-04-25:
You appear to only be using a single wall. Voron spec is 4 walls.

grmocg, 2022-04-25:
Ah yea, I see how that image could confuse-- I have the first layer only use one perimeter. The rest uses the thicker walls.
[image]

Thusfar converting the STL to a body in SW, so I can show what I mean isn't working well.

grmocg, 2022-04-25:
Here is a photoshop edit that sorta shows what I'm poorly communicating:

Here I'm faking putting a void between the perimeters of the outer wall and the inner wall, and approximating what a slicer might do with it.

[image]

grmocg, 2022-04-25:
I'll just close this for now. When the step is out, I'll submit something for the next rev if it works.

closed 2022-04-25 by grmocg

wile-e1, 2022-04-25:
There is an inner profile and an outer profile, how are you suggesting that you make a closed profile without having a start and a stop?

grmocg, 2022-04-25:
It'd still have a start and stop. The hope would be that it'd have one
fewer start+stop, if the inner and outer are connected.
I'll play around with things after the step is out to see if it works on
this.

wile-e1, 2022-04-25:
If you're getting artifacts on your start/stops, it means you need to do some more tuning.  I don't think relying on trickery is a substitute for proper tuning.