VoronDesign/Voron-Afterburner #122: [a]_stealthburner_main_body_beta7.stl hole when slicing.

mlee12382, 2022-04-28:
When I sliced [a]_stealthburner_main_body_beta7.stl using SuperSlicer 2.4.58.2 and using multi color painting I found a hole when inspecting the tool paths, it appears in the upper right mounting hole and only seems to occur when I have it set to the secondary color for contrast. Initially I thought it was a result of a modification I had made but after some testing I was able to reproduce it on the unmodified stl also.
[image]
[image]

wile-e1, 2022-04-29:
> When I sliced [a]_stealthburner_main_body_beta7.stl using SuperSlicer 2.4.58.2 and using multi color painting I found a hole when inspecting the tool paths, it appears in the upper right mounting [...]

I am definitely not seeing this same issue SS 2.3.57 (which in my experience is the last stable version there has been of SS in months).  The screenshot here shows all of the standard Voron settings with the addition of "overlapping perimeters=off" and "thin walls=off".   So if there was a hole....  this would absolutely show it.

[image]

So I'm going to close up this issue, because it is clearly a slicer issue.

closed 2022-04-29 by wile-e1

mlee12382, 2022-04-29:
It only shows up when I "paint" that area using multimaterial painting, which is not available in 2.3.57, I will test it on PrusaSlicer to see if it shows up there also and update.

wile-e1, 2022-04-29:
> It only shows up when I "paint" that area using multimaterial painting, which is not available in 2.3.57, I will test it on PrusaSlicer to see if it shows up there also and update.

The design is not optimized with multi-material printing in mind.  If the part works when you print it with 1 color and it doesn't when you try to do weird slicer stuff to it, that isn't an issue with the STL.

mlee12382, 2022-04-29:
Ok, just thought I would bring it to your attention in case it was something you wanted to address, does the same thing in PrusaSlicer btw. 
[image]
Thanks for your time :)

mlee12382, 2022-04-29:
Any chance you might have an idea of how I can fix it on my end?

wile-e1, 2022-04-29:
I'm guessing it has to do with the fact that there aren't 2 full walls thickness there.  I can probably rework that area to make sure it's atleast 2 walls thick.  Otherwise, I don't have any experience with MM, so I'm not really much help.

mlee12382, 2022-04-29:
Hmm ok, maybe I will try painting the inside of that area the same color and see if it fills the hole if that's the issue. Thanks for your time and suggestion  :)

mlee12382, 2022-04-29:
Well that kinda fixed it but there was 1 tiny line that I couldn't get to fill in, so I will just avoid painting around that hole, thanks again for your help :) Quick question on the dual bowden parts for AB, they should fit sb still correct?