VoronDesign/Voron-Afterburner #125: How to use adxl_washer

XEL-Maker, 2022-04-30:
I printed out the parts needed for adxl 345 mounting. 
There was also a washer_x2, so I printed it, but I couldn't imagine how it would be assembled.
Someone asked for an image on a closed issue related to this, but there has been no comment since.

XEL-Maker, 2022-04-30:
I tried a few more times after writing the issue.
And I assembled it as shown in the picture.
Do you fix the adxl with m2 self tapping?
[image]

kaibsora, 2022-05-03:
you fix it with the supplied washers that are in the same folder, with the m2 self tapping screws. the screws go through the washer, and then self thread into the mount. you cant just use the screws, as the holes are too big, the washers fix that issue.

BeardedTinker, 2022-05-03:
Also, the washers have cutout on one of the surfaces - that part goes over those capacitors near the hole to protect them from both contact and damage if thightening too much.
[image]

XEL-Maker, 2022-05-03:
Thank you for the information.
I'll have to install it as needed.

closed 2022-05-03 by XEL-Maker