VoronDesign/Voron-Afterburner #130: Filament Engagement Issue

Durahl, 2022-05-25:
Greetings...

I was having a busy week with my `Beta0` parts causing me to looe grip which I assume must have had its root in one of the Accent parts keeping tension on the Drive Wheels. The System has an **ERCF** as part of it which is kinda adding _a lot_ of friction along the way the Filament has to go through, but I never had any issues until now.

So, with that I decided to make the jump to the current `Beta7/10/11` Parts since I had to reprint them anyway and have noticed it being practically worse than with the deformed parts. After lot of back and forth I noticed a bunch of issues:

- The Filament Path appears to be unusually large in diameter? I was able to fit a 2.1mm Drill bit into it with ease whereas with the old Afterburner I recall the necessity of boring it to size - Is this supposed to be the case?
- Unlike with the Beta0 Parts where the Filament was grinded equally flat on both sides the Beta7-11 ones would cause only the Passive Gear side ground flat whereas the Active Gear Side would remain pristine.
While feeding a Filament through the open System ( no Motor Plate ) I noticed the Filament making practically no contact with the Active Gear - I can also practically not see the Gear while viewing it from the top through the Filament Path.
Grinding is most definitely not the goal but, in my ( Pamphlet sized ) Book having only one side ground flat is kinda worse, no?
- Without the Active Gear but the Passive Gear Installed I noticed a somewhat similar trend where I was able to pass the Filament through the path with it just barely making any more contact than with the Active Gear.
Bringing the Passive Gear closer is impossible due to the top edge of the Lever resting on a Lip of the Main Body ( above the Ramp ) and the Shuttle also being too long when fully seated.
The Filament **_does_** get properly gripped when both Gears engage it but apparently not enough under certain situations with all the ERCF Friction going on along ( influx of humidity in recent days? Dunno... ), the Active Gear not being situated in the Filament Path and the Passive Gear ( or rather the Lever it is mounted to ) bottoming out before getting a secure hold onto the Filament.
( My biggest Suspect for the Friction Problem also being the Entry / Exit holes of the ERCP Buffer Zone as they're not perpendicular to the Wheel the Filament is getting spooled onto - [Especially not the Exit one feeding to the ERCF](https://i.imgur.com/et7Aa7A.png). )

While I'm at it some questions that have also popped up ever since making the change to the Stealthburner:
1. The 10T Stepper Motor Gear is only making Contact with about ⅔ of the BMG Plastic Gear ( back side ) leaving about ⅓ ( front side ) untouched despite the Axle making contact with the Stepper Housing ( visible grinding marks ). Are we supposed to grind the Axle down in such a case to both improve the engagement between the Gears and to prevent the Axle from making contact with the Stepper Housing? At the same time I notice the Axle already not being proud in a properly seated front Bearing kinda indicating that this one too would have to be shifted further inwards for the axle to still be making a reasonable amount of contact with the inside of the Bearing? 🤔
2. May I suggest the addition of a third cooling channel from the Hotend Cooling Fan towards the inside of the CW2 like how it was done for the bottom of the Part Cooling Shroud? I cannot shake the idea that it may aid in preventing like the Levers and Bearing Retainers from deforming while under constant tension while getting heat soaked from the Hotend and Stepper Motor?
Cheers 😁
3. Is there a chance for an early access to the CAD Solids? I've since done okay modifying the STLs via Modifiers in the Slicer to suite my needs but realigning the Bearings to fix the Active Gear Filament Engagement issue is beyond tedious doing it like that 😑

xinqi9083, 2022-06-29:
Same problem,  I had to replace my CW2 with Orbiter 2.0

raymondh2, 2022-07-26:
Closing as beta has ended

closed 2022-07-26 by raymondh2