VoronDesign/Voron-Afterburner #132: Clockwork2 lack of filament grip

bartoszwas, 2022-06-17:
Hey,

I've printed the last beta Clockwork 2. Issue is that I got lack of filament grip. When I put filament in and close the latch I can pull it out without moving the gears. Teeth leave almost no sign on filament so its skipping and I got under-extrusion.

nemesina, 2022-06-17:
Same problem here. With last stl's.

Clarkbf, 2022-06-17:
I had the same problem. Went back to CW1 because I did not see anyone else raise the issue. Usually with something like this it is user error so I didn't bring it up, even though I tried many different ways to get it to work.

hmouridsen, 2022-06-17:
From my experience the latch shuttle needs some exercise - making is moving freely, and gain the tension on the gears